google/aiyprojects-raspbian #655: Running `image_classification.py` in Docker Container results in Host connection refused error.

MayankTahil, 2019-11-30:
# Overview

Trying to set up a way to test some of the code to classify images without the need for the AIY kit physically. Will not be relying on the camera module but instead using sample images to run tests to validate outputs. Creating a Docker image with dependencies pre-bundled that can be recycled for CI jobs during commits and development

# References 

These were helpful links to help me get started : 

* [Install the AIY Projects software](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md) for instructions when installing on a clean Raspbian SD card.
* [uglywan/aiy-voice-kit-docker](https://hub.docker.com/r/uglywan/aiy-voice-kit-docker/dockerfile) for reference on previous work on docker images for AIY kits. 
* [google/aiyprojects-raspbian](https://github.com/google/aiyprojects-raspbian) home project repository

# Problems

Refer to [Dockerfile](https://gitlab.com/mayanktahil/rosie/blob/master/dockerfile) to replicate environment 

When in the `/app` directory, issuing the command: 

`./image_classification.py -i ./test_images/foam_cup.jpg` creates the following error: 

```bash
Traceback (most recent call last):
  File "./image_classification.py", line 67, in <module>
    main()
  File "./image_classification.py", line 47, in main
    with ImageInference(image_classification.model(model_type)) as inference:
  File "/app/aiy/vision/inference.py", line 152, in __init__
    self._engine = self._stack.enter_context(InferenceEngine())
  File "/app/aiy/vision/inference.py", line 275, in __init__
    self._transport = make_transport()
  File "/app/aiy/vision/_transport.py", line 87, in make_transport
    return _SocketTransport()
  File "/app/aiy/vision/_transport.py", line 70, in __init__
    self._client.connect((host, port))
ConnectionRefusedError: [Errno 111] Connection refused
```

I can narrow it down to the following [file and line(s) of code](https://gitlab.com/mayanktahil/rosie/blob/master/aiy/vision/_transport.py#L70): 

```python
class _SocketTransport:
    """Communicate with VisionBonnet over socket."""

    def __init__(self):
        """Open connection to the bonnet."""
        self._client = socket.socket(socket.AF_INET, socket.SOCK_STREAM)

        host = os.environ.get('VISION_BONNET_HOST', '172.28.28.10')
        port = int(os.environ.get('VISION_BONNET_PORT', '35000'))
        self._client.connect((host, port))
```

The fact the connection is refused makes me think that it's a networking issue caused by the fact I'm running this code in a Docker container and issuing websocket connections from an overlay network. Can't figure a way to test this out in a replicable, isolated environment without a dedicated linux host and without the AIY kit hardware.

MayankTahil, 2019-11-30:
Tried to replicate dockerfile instructions on a GCP Ubuntu host as well which produced same error / issue.

dmitriykovalev, 2019-12-03:
You cannot test image classification (and any other ML algorithm) without Vision Bonnet board physically connected to the Pi. The board contains a chip which does all ML computations.

MayankTahil, 2019-12-03:
That explains a lot. I realized this just last night and should have updated this issue accordingly. Thanks @dmitriykovalev .

closed 2019-12-03 by dmitriykovalev