google/aiyprojects-raspbian #656: Pi 3B not recognizing voice HAT with Buster and hacking.md info

AmorOdium, 2019-12-01:
Hello all, I have followed all the steps in the below URL twice now (I did a clean install and started over after it failed the first time. 

https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

I can get everything to work without errors but when I do the audio test recording/replay I get nothing. I also have found after disabling internal audio, the device does not detect the AIY HAT so I believe that is the core of the issue. 

I have successfully installed the drivers according to the guide, and added the HAT overlay to the boot config file (as instructed by a forum post somewhere else), even after a reboot, I am not getting much. Not sure where to go from here as these devices don't seem to be very well documented as far as forum posts and troubleshooting.

AmorOdium, 2019-12-01:
Notes to add, I am using the Classic Voice kit, not the one for the pi zeroW. I just bought it on black friday without realizing it was such an old device. I'm using a rev1.2 board. and I have also now tried swapping to another pi with the same results. So it seems this is software related or the Voice HAT is bad possibly?

manoj7410, 2019-12-04:
Can you share the output of the command : 

- lsmod

AmorOdium, 2019-12-04:
lsmod
Module                  Size  Used by
fuse                  110592  3
rfcomm                 49152  6
bnep                   20480  2
hci_uart               40960  1
btbcm                  16384  1 hci_uart
serdev                 20480  1 hci_uart
bluetooth             389120  29 hci_uart,bnep,btbcm,rfcomm
ecdh_generic           28672  1 bluetooth
8021q                  32768  0
garp                   16384  1 8021q
stp                    16384  1 garp
llc                    16384  2 garp,stp
spidev                 20480  0
brcmfmac              311296  0
brcmutil               16384  1 brcmfmac
sha256_generic         20480  0
cfg80211              614400  1 brcmfmac
rfkill                 28672  6 bluetooth,cfg80211
raspberrypi_hwmon      16384  0
hwmon                  16384  1 raspberrypi_hwmon
snd_soc_rpi_simple_soundcard    16384  1
i2c_bcm2835            16384  0
bcm2835_v4l2           45056  0
bcm2835_codec          36864  0
v4l2_common            16384  1 bcm2835_v4l2
v4l2_mem2mem           24576  1 bcm2835_codec
bcm2835_mmal_vchiq     32768  2 bcm2835_codec,bcm2835_v4l2
spi_bcm2835            20480  0
videobuf2_vmalloc      16384  1 bcm2835_v4l2
videobuf2_dma_contig    20480  1 bcm2835_codec
videobuf2_memops       16384  2 videobuf2_dma_contig,videobuf2_vmalloc
videobuf2_v4l2         24576  3 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem
videobuf2_common       45056  4 bcm2835_codec,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_soc_bcm2835_i2s    20480  2
snd_soc_googlevoicehat_codec    16384  1
videodev              200704  6 bcm2835_codec,v4l2_common,videobuf2_common,bcm2835_v4l2,v4l2_mem2mem,videobuf2_v4l2
snd_soc_core          192512  3 snd_soc_googlevoicehat_codec,snd_soc_bcm2835_i2s,snd_soc_rpi_simple_soundcard
media                  36864  3 bcm2835_codec,videodev,v4l2_mem2mem
vc_sm_cma              36864  1 bcm2835_mmal_vchiq
snd_compress           20480  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
snd_pcm               102400  4 snd_pcm_dmaengine,snd_soc_bcm2835_i2s,snd_soc_core,snd_soc_rpi_simple_soundcard
snd_timer              32768  1 snd_pcm
snd                    73728  6 snd_compress,snd_timer,snd_soc_core,snd_pcm
uio_pdrv_genirq        16384  0
uio                    20480  1 uio_pdrv_genirq
fixed                  16384  0
i2c_dev                20480  0
ip_tables              24576  0
x_tables               32768  1 ip_tables
ipv6                  450560  32

AmorOdium, 2019-12-07:
@manoj7410  have you had a chance to take a look at my output/do you have any suggestions as to what to try?

manoj7410, 2019-12-07:
@AmorOdium , to rule out any hardware issues, can you try to install the image available at https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113. This image has worked for me on Voice HAT with RPI 3B and 3B+. 
Meanwhile, I'll also try to work with HACKING.md again and will share the results in next week.

AmorOdium, 2019-12-07:
seems to work fine on this image Manoj, the test script runs without issue (after some updates at least)

AmorOdium, 2020-01-04:
Hey Manoj, did you ever get a chance to take a look at this?

viraniac, 2020-01-07:
@AmorOdium I tried configuring my raspian to see if I can get it to work after seeing your comment. As expected, it didn't worked with just the instructions provided here.

In order to get it working, please check the following
1) I2C should be on in rpi-config or Preferences > Raspberry Pi configuration
2) The following line should be present in boot/config.txt for voicehat to work:
dtparam=i2s=on
dtoverlay=i2s-mmap
dtoverlay=googlevoicehat-soundcard

3) You might also have to use this [~/.asoundrc](https://raw.githubusercontent.com/shivasiddharth/GassistPi/master/audio-drivers/AIY-HAT/scripts/.asoundrc) file

I had long back faced these issues and found this solution on a reddit thread. The credit goes to original reddit post creator.