google/aiyprojects-raspbian #657: assistant_library_demo.py Error in the assistant.py

kkarpa, 2019-12-01:
When I try and run the assistant_library_demo.py I run into a Segmentation fault when the program gets into the assistant.py, specifically when it is trying to encode into ASCII. I also get this problem in when I'm trying to run assistant_library_with_button_demo.py but not in assistant_grpc_demo.py. I was wondering if there's just an error in the code and its an easy fix or if I need to re-flash the program?

manoj7410, 2019-12-04:
Assistant library has been deprecated by Google assistant team and isn't supported anymore.

screamnAbdab, 2019-12-09:
@kkarpa , I ran into the same issue. Got frustrated. Installed mycroft (https://mycroft-ai.gitbook.io/docs/using-mycroft-ai/get-mycroft/picroft) and now I have a nice assistant that is easily hackable/extendable. It is not as polished as google's offering but it is open source and has great potential.

odechr, 2019-12-22:
The kit works okey, only issue is that you HAVE to push the button to activate the assistant. Is this related to the same issue regarding Assistant library is deprecated?

I mean, only issue i got is that i will not respond to just normal "OK google" @dmitriykovalev?

Im all new too this.

workingmanrob, 2020-01-01:
`pip3 install google-assistant-library==1.0.0`
`pip3 install google-assistant-library`

Try running that.  Worked for me after I ran that as the pi user.

yotastalker, 2020-01-04:
[image]

Above screen capture on my pi.

I've been trying to resolve the No such file or directory issue for a week or so between work and having a life. I looked in the folder /examples and it is there but some reason it still says No such file. 

Any ideas to go from here.

odechr, 2020-01-05:
> [image]
> 
> Above screen capture on my pi.
> 
> I've been trying to resolve the No such file or directory issue for a week or so between work and having a life. I looked in the folder /examples and it is there but some reason it still says No such file.
> 
> Any ideas to go from here.

assistant_library_demo.py is not a folder, so you cannot "cd" into it. Inside the "voice" folder its only files.

odechr, 2020-01-05:
> `pip3 install google-assistant-library==1.0.0`
> `pip3 install google-assistant-library`
> 
> Try running that. Worked for me after I ran that as the pi user.

Wow, that actually worked! Im not even sure what it did, since the library is not supported anymore how can we still use it?

workingmanrob, 2020-01-07:
https://github.com/google/aiyprojects-raspbian/issues/635#issuecomment-524366391

My response is right at the end of that post... just before the bit in bold.

Glad to hear it worked for you too though!