google/aiyprojects-raspbian #658: Error when executing src/examples/voice/assistant_grpc_demo.py

vishwapinnawala, 2019-12-12:
newbie here Help!!! Tried Re enabling Assistant API and i tried installing AIY os again and again 3 times no luck 

Traceback (most recent call last):
  File "src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "src/examples/voice/assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
    super().__init__(language_code, volume_percentage)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given

manoj7410, 2019-12-13:
@vishwapinnawala Which version of voice kit do you have ?
Have you reflashed the kit with the image provided at https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 ?

vishwapinnawala, 2019-12-14:
V2 yes i tried reflashing several times and nothing seems to work

daOrangePeeler, 2019-12-23:
I'm getting an identical error with voice kit v1. I also tried reflashing the image with no success.

daOrangePeeler, 2019-12-24:
I think I fixed the error for my setup with the Raspberry Pi 3b with AIY Voice Kit v1. In the file "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py",  I changed the last argument in webbrowser.register() (line 75) from "-1" to "preferred=True"

Here is the [documentation](https://docs.python.org/3/library/webbrowser.html#webbrowser.register) that explains the options. It looks like in Python 3.7, the webbrowser.register() preferred argument only takes keywords now.

Here is the [full, edited file](https://drive.google.com/file/d/0B8AGLT2g1qZ2aXpOeFdGbTJRU1dzY3c0R0VuWDFsZXNibzdR/view?usp=sharing) as well. GitHub wouldn't let me upload the file directly, so the file is on Google Drive.

Or, if you just want to edit the code directly, here is the new code for line 75 in the file "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py":
`webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), preferred=True)`

Please let me know if it works or not, or if you have any questions.

stevesss, 2019-12-26:
The above solved the problem of the browser not authenticating the voice kit, but it just keeps opening the browser over and over again and never starts working (like it used to when i installed it before)

vishwapinnawala, 2019-12-27:
i've removed and got alexapi installed so i gived it up😁

daOrangePeeler, 2019-12-27:
> The above solved the problem of opening the browser and authenticating the voice kit as it takes me to the web page that says the process is complete, but then the next time I do "assistant_grpc_demo.py" again, it just starts this authentication process all over again instead of starting up like it is supposed to. (I had it working before.) I am stuck at this point for now.

I'm sorry to hear that stevesss. I just reflashed the image, created a new Google Cloud project and edited the auth_helpers.py file without success in replicating your problem. What Raspberry Pi and Voice Kit version are you using? I know it is a long process, but you might want to try reflashing your image.

d3acon, 2019-12-29:
Editing auth_helpers.py solved it for me too. But I get a seg fault, once I did the authorisation in the browser.

mcmanus-git, 2019-12-30:
I'm having the same issue as @d3acon

daOrangePeeler, 2019-12-30:
I get a segmentation fault as well for every file except "assistant_grpc_demo.py"

d3acon, 2019-12-31:
I opened a new ticket for the seg fault, as this one is for the python problem.

2000prasanth, 2020-02-06:
please check i have written a blog on all error that i have encountered
https://2000prasanth.wixsite.com/website/blog/aiy-voice-how-i-set-up-my-voice-kit-amid-errors-my-first-raspberry-pi-project

downeyjar, 2020-02-13:
Traceback (most recent call last):
  File "/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "/home/pi/AIY-voice-kit-python/src/examples/voice/assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
    super().__init__(language_code, volume_percentage)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given

I am using raspberry pi 4

2000prasanth, 2020-02-13:
i had the same error
pleasw try this

I fixed the error for my setup with the Raspberry Pi 3B with AIY Voice Kit
v1. In the file
"/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", I changed
the last argument in webbrowser.register() (line 75) from "-1" to
"preferred=True".

Here is the documentation
<https://docs.python.org/3/library/webbrowser.html#webbrowser.register>
that explains the options for webbrowser.register(). It looks like in
Python 3.7, the webbrowser.register() "preferred" argument only takes
keywords now.

 edit the code directly, here is the new code for line 75 in the file
"/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py":

webbrowser.register('chromium-browser', None,
webbrowser.Chrome('chromium-browser'), preferred=True)

(Make sure to leave the spaces from the original line of code in front of
the new code segment as spacing matters in Python).

downeyjar, 2020-02-16:
i have raspberry pi 4