google/aiyprojects-raspbian #659: Vision kit: problems with boot, sd card and connnection.

emilio-calvo, 2019-12-22:
Hi. I'm starting with my new vision kit and i have some problems.

Firstly, when i boot up the vision kit i not hear any beep. I can see the gren led, but the push button start to shine with a intermittent red colour. I have try with differetnts usb power supply, between 2.0 and 2.5 Amp, all with same result.

Secondly, i can't flash the new image on the sd card because my laptop don't recognize it.

And finally i try to connect my vision kit by AIY app and my laptop, but when i used the Secure Shell Extension in Chrome, appears "port 22: connection refused. NaCl complement has been closed with status code 225"

Can someone help me?

emilio-calvo, 2019-12-22:
> Hi. I'm starting with my new vision kit and i have some problems.
>
> Firstly, when i boot up the vision kit i not hear any beep. I can see the gren led, but the push button start to shine with a [...]

manoj7410, 2020-02-06:
The first issue : "Blinking red LED" indicates that joy_demo has crashed. Most probably due to camera connection being loose.

closed 2020-11-23 by dmitriykovalev