google/aiyprojects-raspbian #660: Segmentation fault with assistant_library_demo.py

d3acon, 2019-12-29:
I am running the AIY Kit V1 with a Raspberry Pi 3 Model B Rev 1.2 and debian 10.2 (just ran the update after flashing aiyprojects-2019-11-13.img.xz).
After allowing the google assistant api to access my google account in the browser, the demo apps quit with a seg fault.
pi@aiy:~/AIY-projects-python/src/examples/voice $ ./assistant_library_demo.py 
Segmentation fault
Did I miss something? Thank you!

pythonmoti, 2020-01-01:
> I am running the AIY Kit V1 with a Raspberry Pi 3 Model B Rev 1.2 and debian 10.2 (just ran the update after flashing aiyprojects-2019-11-13.img.xz).
> After allowing the google assistant api to [...]

I just had the same issue. so I decided to try the original "VoiceKit Classic Image 2017-09-11" and it thank G-d worked perfectly . happy to help

workingmanrob, 2020-01-01:
`pip3 install google-assistant-library==1.0.0`
`pip3 install google-assistant-library`

Using the image from 2019-11-13 I needed to run those commands as the pi user or the assistant_library_demo.py would just seg fault

rafael-oropeza, 2020-01-10:
> `pip3 install google-assistant-library==1.0.0`
> `pip3 install google-assistant-library`
> 
> Using the image from 2019-11-13 I needed to run those commands as the pi user or the assistant_library_demo.py would just seg fault

Hi

Even running pip3 commands shown above I'm getting "seg-fault" error"

`pi@raspberrypi:~ $ sudo pip3 install google-assistant-library==1.0.1
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: google-assistant-library==1.0.1 in /usr/local/lib/python3.7/dist-packages-linux_armv7l (1.0.1)
Requirement already satisfied: requests<3,>=2.14.2 in /usr/lib/python3/dist-packages (from google-assistant-library==1.0.1) (2.21.0)
Collecting argparse<2,>=1.4.0 (from google-assistant-library==1.0.1)
  Downloading https://files.pythonhosted.org/packages/f2/94/3af39d34be01a24a6e65433d19e107099374224905f1e0cc6bbe1fd22a2f/argparse-1.4.0-py2.py3-none-any.whl
Requirement already satisfied: enum34<2,>=1.1.6 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library==1.0.1) (1.1.6)
Requirement already satisfied: pathlib2<3,>=2.3.0 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library==1.0.1) (2.3.5)
Requirement already satisfied: google-auth<2,>=1.0.1 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library==1.0.1) (1.6.3)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from pathlib2<3,>=2.3.0->google-assistant-library==1.0.1) (1.12.0)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library==1.0.1) (0.2.7)
Requirement already satisfied: cachetools>=2.0.0 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library==1.0.1) (3.1.1)
Requirement already satisfied: rsa>=3.1.4 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library==1.0.1) (4.0)
Requirement already satisfied: pyasn1<0.5.0,>=0.4.6 in /usr/local/lib/python3.7/dist-packages (from pyasn1-modules>=0.2.1->google-auth<2,>=1.0.1->google-assistant-library==1.0.1) (0.4.7)
Installing collected packages: argparse
Successfully installed argparse-1.4.0
pi@raspberrypi:~ $ sudo pip3 install google-assistant-library
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: google-assistant-library in /usr/local/lib/python3.7/dist-packages-linux_armv7l (1.0.1)
Requirement already satisfied: requests<3,>=2.14.2 in /usr/lib/python3/dist-packages (from google-assistant-library) (2.21.0)
Requirement already satisfied: enum34<2,>=1.1.6 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library) (1.1.6)
Requirement already satisfied: google-auth<2,>=1.0.1 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library) (1.6.3)
Requirement already satisfied: pathlib2<3,>=2.3.0 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library) (2.3.5)
Requirement already satisfied: argparse<2,>=1.4.0 in /usr/local/lib/python3.7/dist-packages (from google-assistant-library) (1.4.0)
Requirement already satisfied: pyasn1-modules>=0.2.1 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library) (0.2.7)
Requirement already satisfied: rsa>=3.1.4 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library) (4.0)
Requirement already satisfied: six>=1.9.0 in /usr/lib/python3/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library) (1.12.0)
Requirement already satisfied: cachetools>=2.0.0 in /usr/local/lib/python3.7/dist-packages (from google-auth<2,>=1.0.1->google-assistant-library) (3.1.1)
Requirement already satisfied: pyasn1<0.5.0,>=0.4.6 in /usr/local/lib/python3.7/dist-packages (from pyasn1-modules>=0.2.1->google-auth<2,>=1.0.1->google-assistant-library) (0.4.7)

pi@raspberrypi:~ $ /home/pi/AIY-projects-python/src/examples/voice/assistant_library_with_local_commands_demo.py
Segmentation fault`

I'm stuck, :-(

robotastic, 2020-01-15:
I have a V1 Voice Kit and installing the earlier version of the library fixed it for me.

pythonmoti, 2020-01-15:
Which version exactly?

robotastic, 2020-01-15:
I used this command: 
`pip3 install google-assistant-library==1.0.0`

pythonmoti, 2020-01-15:
Thank You very much.
From: Luke Berndt
Sent: Tuesday, January 14, 2020 9:19 PM
To: google/aiyprojects-raspbian
Cc: pythonmoti; Comment
Subject: Re: [google/aiyprojects-raspbian] Segmentation fault withassistant_library_demo.py (#660)

I used this command:
pip3 install google-assistant-library==1.0.0

fumantsu, 2020-02-01:
I have the same and solve it with the older version. It is kind sucks that "support" for the V1 version has been "stopped"

mdbrewer, 2020-03-13:
Check out my writeup on how to get it working and update stretch. https://medium.com/@michaeldbrewer/aiy-voice-deprecation-workaround-bd07559403df

saurabhsharma001, 2020-05-17:
`pip3 install google-assistant-library==1.0.0`
resolved the problem.