google/aiyprojects-raspbian #662: Use as posture detector?

Tylersuard, 2020-01-02:
Hello!  I love the AIY kit, and it's worked perfectly as a smile detector.  I'm wondering how I can program it to become a correct posture detector?  There are models online that detect a person's estimated limb positions, neck, head, etc.  Can I reprogram my AIY to do that limb position estimation, and to alert me when my posture is sub-optimal?

manoj7410, 2020-02-19:
This can be useful for you https://www.hackster.io/dvillevald/hand-command-recognizer-on-google-aiy-vision-kit-3786f6. This is not exactly posture detection but somehow related to it. I haven't tested this demo though.

Tylersuard, 2020-02-19:
This is perfect, thank you!

dmitriykovalev, 2020-11-20:
Please also check https://github.com/google-coral/project-posenet.

closed 2020-11-20 by dmitriykovalev