google/aiyprojects-raspbian #672: Vision bonnet blocks GPIO pins 14 & 15. How can I connect to these?

webxl, 2020-02-13:
I was thinking about soldering wires to the back side of the pins on the bonnet but would that cause any issues? They appear to be unused: https://pinout.xyz/pinout/aiy_vision_bonnet

There's also the possibility of creating another UART interface using the expansion GPIO pins, but that seems really tricky: https://raspberrypi.stackexchange.com/questions/24019/can-the-gpios-pins-be-used-as-another-rx-and-tx/24039

Hopefully the maintainers can figure out a solution or the product people can provide access in future versions of the hardware. Thanks!

manoj7410, 2020-02-19:
Is there a specific requirement of soldering the wires on the back side of Bonnet ? You can solder on the back side of the PI too right ?

dmitriykovalev, 2020-11-20:
That's correct, the only current way to use 'unused' GPIOs is to solder on either Pi or Bonnet side.

closed 2020-11-20 by dmitriykovalev