google/aiyprojects-raspbian #679: Visual Response

Redblock6YT, 2020-03-13:
Hello,
I was wondering how I could implement the "Visual Response" feature that is in the Google Assistant SDK. 

If you do not know what I am referring to, you can click the link below:
https://developers.google.com/assistant/sdk/guides/service/python/embed/next-steps

manoj7410, 2020-05-12:
Have you tried to follow the instructions given at : https://github.com/googlesamples/assistant-sdk-python/tree/master/google-assistant-sdk/googlesamples/assistant/grpc#python-samples-for-the-google-assistant-grpc-api and https://github.com/googlesamples/assistant-sdk-python/blob/master/google-assistant-sdk/googlesamples/assistant/grpc/textinput.py#L149

Redblock6YT, 2020-06-15:
I tried, but with the AIY, it automatically starts when the Rasp PI starts up. I would need more information in order to continue.

manoj7410, 2020-06-16:
> I tried, but with the AIY, it automatically starts when the Rasp PI starts up. I would need more information in order to continue.

What does automatically start ?

dmitriykovalev, 2020-11-20:
Please follow the instructions from https://github.com/google/aiyprojects-raspbian/issues/679#issuecomment-627128514.

closed 2020-11-20 by dmitriykovalev