google/aiyprojects-raspbian #680: assistant_library only works with ethernet

jmrocco, 2020-04-03:
I'm a beginner, I've been having a heck of a time trying to get this V1 Voice Kit to work. The 2019-11-13 release wasn't working for me so I switched to the 2018-11-16 release. Had lots of issues connecting to the WIFI. It just wouldn't connect. I tried multiple solutions on the internet. Finally it just connected one night out of the blue. I then came across the Google Assistant depreciation so I installed an earlier version with `pip3 install google-assistant-library==1.0.0`. When I try to run the assistant_library_demo.py script, or any of the scripts for that matter, I get the following errors.
`Traceback (most recent call last):
  File "./assistant_library_demo.py", line 66, in <module>
    main()
  File "./assistant_library_demo.py", line 59, in main
    credentials = auth_helpers.get_assistant_credentials()
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 132, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 100, in _try_to_get_credentials
    return _load_credentials(_ASSISTANT_CREDENTIALS)
  File "/opt/aiy/projects-python/src/aiy/assistant/auth_helpers.py", line 63, in _load_credentials
    credentials.refresh(http_request)
  File "/home/pi/.local/lib/python3.5/site-packages/google/oauth2/credentials.py", line 183, in refresh
    self._scopes,
  File "/home/pi/.local/lib/python3.5/site-packages/google/oauth2/_client.py", line 248, in refresh_grant
    response_data = _token_endpoint_request(request, token_uri, body)
  File "/home/pi/.local/lib/python3.5/site-packages/google/oauth2/_client.py", line 105, in _token_endpoint_request
    response = request(method="POST", url=token_uri, headers=headers, body=body)
  File "/home/pi/.local/lib/python3.5/site-packages/google/auth/transport/requests.py", line 181, in __call__
    method, url, data=body, headers=headers, timeout=timeout, **kwargs
  File "/home/pi/.local/lib/python3.5/site-packages/requests/sessions.py", line 530, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/requests/sessions.py", line 643, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/requests/adapters.py", line 449, in send
    timeout=timeout
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/connectionpool.py", line 672, in urlopen
    chunked=chunked,
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/connectionpool.py", line 376, in _make_request
    self._validate_conn(conn)
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/connectionpool.py", line 994, in _validate_conn
    conn.connect()
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/connection.py", line 300, in connect
    conn = self._new_conn()
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/connection.py", line 157, in _new_conn
    (self._dns_host, self.port), self.timeout, **extra_kw
  File "/home/pi/.local/lib/python3.5/site-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)`

I then moved locations in the house and connected to ethernet. Now it works perfect. As soon as I disconnect from ethernet, not even moving locations, It doesn't work. I have downloaded everything correctly as far as I can tell.

manoj7410, 2020-05-12:
Is this issue still reproducible ?

dmitriykovalev, 2020-11-20:
Our SD card image doesn't change any networking options, both Ethernet and Wi-Fi should work the same way if configured properly.

closed 2020-11-20 by dmitriykovalev