google/aiyprojects-raspbian #681: Problemas con el vision kit AIY

ibrunoes, 2020-04-12:
Hola a todos, después de construir el vision kit y conectarlo por SSH va todo perfecto pero cuando intento cargar un programa python de la camara me indica este error:

pi@ibruno:~/AIY-projects-python/src/examples/vision $ ./image_classification_camera.py
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
Traceback (most recent call last):
File "/usr/lib/python3/dist-packages/picamera/camera.py", line 456, in _init_camera
self._camera = mo.MMALCamera()
File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 2279, in init
super(MMALCamera, self).init()
File "/usr/lib/python3/dist-packages/picamera/mmalobj.py", line 633, in init
prefix="Failed to create MMAL component %s" % self.component_type)
File "/usr/lib/python3/dist-packages/picamera/exc.py", line 184, in mmal_check
raise PiCameraMMALError(status, prefix)
picamera.exc.PiCameraMMALError: Failed to create MMAL component b'vc.ril.camera': Out of memory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "./image_classification_camera.py", line 64, in
main()
File "./image_classification_camera.py", line 54, in main
with PiCamera(sensor_mode=4, framerate=30) as camera,
File "/usr/lib/python3/dist-packages/picamera/camera.py", line 431, in init
self._init_camera(camera_num, stereo_mode, stereo_decimate)
File "/usr/lib/python3/dist-packages/picamera/camera.py", line 460, in _init_camera
"Camera is not enabled. Try running 'sudo raspi-config' "
picamera.exc.PiCameraError: Camera is not enabled. Try running 'sudo raspi-config' and ensure that the camera has been enabled.

La camara está en enable y no la reconoce.
Alguien me puede ayudar por favor??

manoj7410, 2020-05-06:
Is this issue still reproducible ? ENOMEM generally says no memory left on the device.

ibrunoes, 2020-05-28:
Sigue el problema, le cambie la tarjeta de 8gb que venía con el kit por otra de 16gb y continua igual

ibrunoes, 2020-05-28:
si alguien puede ayudarme le estaría muy agradecido, ahi la tengo en la cajita de carton totalmente inservible. que pena.

ricardodeazambuja, 2020-05-29:
@ibrunoes Have a look at this discussion:
https://raspberrypi.stackexchange.com/questions/83630/failed-to-create-mmal-component-bvc-ril-camera-out-of-memory
You may have two processes trying to use the camera at the same time.

Another suggestion is to connect the camera directly to the RPI Zero and test it. I had problems with my AIY Vision Bonnet where the camera would not work directly connected to it anymore and I solved by connecting the camera directly to the RPI.

dmitriykovalev, 2020-11-20:
Closing — this doesn't seem to be related to the Vision Bonnet.

closed 2020-11-20 by dmitriykovalev