google/aiyprojects-raspbian #684: documentation for pwm-aiy-io module

combs, 2020-04-18:
Hello! I am learning through trial and error that the AIY Vision Bonnet's pins have different permissible values for the PWM frequency, at least as set through the Python AIY libraries. 

I can set PIN_A. PIN_C, PIN_D to 1000hz PWM but PIN_B only accepts 100hz. And it seems like particular values are permissible and other values are not, e.g. 2000hz. 

Is there documentation for the pwm-aiy-io module? I searched this repository and looked at Google's other AIY projects repos, and couldn't find anything.

Example of bad value's error output:

```
OSError: [Errno 22] Invalid argument

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "metertest.py", line 12, in <module>
    ledb = PWMLED(PIN_B, frequency=1000)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 418, in __init__
    self.pin.frequency = frequency
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 358, in <lambda>
    lambda self, value: self._set_frequency(value),
  File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 600, in _set_frequency
    self.pwm_pin.set_period_ns(_NS_PER_SECOND / frequency)
  File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 358, in set_period_ns
    self._update_state(new_state)
  File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 322, in _update_state
    self._set_period_ns(new_state.period_ns)
  File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 304, in _set_period_ns
    self.write_property('period', '%d' % period_ns)
  File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 179, in write_property
    node.write(value)
OSError: [Errno 22] Invalid argument

```

combs, 2020-04-18:
and here's the code, for reference:

```
from gpiozero import PWMLED 
from aiy.pins import PIN_A, PIN_B

a = PWMLED(PIN_A, frequency=1000) #works
b = PWMLED(PIN_B, frequency=1000) #fails
b = PWMLED(PIN_B, frequency=100)  #works

```

combs, 2020-04-18:
I think I had it open in two processes. Would still love some documentation for pwm-aiy-io module! Thanks!

ricardodeazambuja, 2020-05-29:
@combs Would you mind sharing the results of your tests? Thanks!

combs, 2020-05-29:
> @combs Would you mind sharing the results of your tests? Thanks!

I had the pin still open in another process, which kept the frequency fixed, so this was a procedural mistake on my part. It went away after a restart.

ricardodeazambuja, 2020-05-29:
I'm curious about the max frequencies and any extra caveat :)

combs, 2020-05-29:
I was getting terrible results because it seems to reset the channel and its frequency every time the PWM value is changed, so any sort of smoothly changing value is broken up with lots of gaps. I discontinued use of the AIY PWM in my application.

The source for the PWM module is on the filesystem in the AIY Raspbian image, though not in any public source control that I could find. It should have the max frequencies in its logic. 

The module is still largely undocumented. If Google used public source control for it, this might be more easily fixed.