google/aiyprojects-raspbian #685: Error in Training embedded_ssd_mobilenet_v1_coco.config

krismanuelp, 2020-05-02:
Hey there! I'm trying to train embedded_ssd_mobilenet_v1_coco.config from with Tensorflow using tutorial by EdjeElectronics and fine_tune_checkpoints by zhoujustin (all links down below) in order to fit 256x256 and depth_multiplier = 0.125 required by AIY Vision Kit. But it produces the following error:

D:\ObjectDetection\models\research\object_detection>python train.py --logtostderr --train_dir=training/ --pipeline_config_path=training/embedded_ssd_mobilenet_v1_coco.config
WARNING:tensorflow:From D:\ObjectDetection\models\research\object_detection\trainer.py:210: create_global_step (from tensorflow.contrib.framework.python.ops.variables) is deprecated and will be removed in a future version.
Instructions for updating:
Please switch to tf.train.create_global_step
INFO:tensorflow:depth of additional conv before box predictor: 0
INFO:tensorflow:depth of additional conv before box predictor: 0
INFO:tensorflow:depth of additional conv before box predictor: 0
INFO:tensorflow:depth of additional conv before box predictor: 0
INFO:tensorflow:depth of additional conv before box predictor: 0
Traceback (most recent call last):
  File "train.py", line 163, in <module>
    tf.app.run()
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python36\lib\site-packages\tensorflow\python\platform\app.py", line 124, in run
    _sys.exit(main(argv))
  File "train.py", line 159, in main
    worker_job_name, is_chief, FLAGS.train_dir)
  File "D:\ObjectDetection\models\research\object_detection\trainer.py", line 228, in train
    clones = model_deploy.create_clones(deploy_config, model_fn, [input_queue])
  File "D:\ObjectDetection\models\research\slim\deployment\model_deploy.py", line 193, in create_clones
    outputs = model_fn(*args, **kwargs)
  File "D:\ObjectDetection\models\research\object_detection\trainer.py", line 167, in _create_losses
    losses_dict = detection_model.loss(prediction_dict)
  File "D:\ObjectDetection\models\research\object_detection\meta_architectures\ssd_meta_arch.py", line 474, in loss
    location_losses, cls_losses, prediction_dict, match_list)
  File "D:\ObjectDetection\models\research\object_detection\meta_architectures\ssd_meta_arch.py", line 640, in _apply_hard_mining
    match_list=match_list)
  File "D:\ObjectDetection\models\research\object_detection\core\losses.py", line 515, in __call__
    location_losses = tf.unstack(location_losses)
  File "C:\Users\Asus\AppData\Local\Programs\Python\Python36\lib\site-packages\tensorflow\python\ops\array_ops.py", line 1054, in unstack
    (axis, -value_shape.ndims, value_shape.ndims))
ValueError: axis = 0 not in [0, 0)

Can you guys help me? Thanks.

Links
Config files: https://github.com/tensorflow/models/blob/master/research/object_detection/samples/configs/embedded_ssd_mobilenet_v1_coco.config
Tensorflow tutorial by EdjeElectronics: https://github.com/EdjeElectronics/TensorFlow-Object-Detection-API-Tutorial-Train-Multiple-Objects-Windows-10
fine_tune_checkpoint by zhoujustin: https://drive.google.com/file/d/1_MeZ8kvmpNibPZvSJGnwKNRATeuyxNtu/view

manoj7410, 2020-05-12:
Which version of TF are you working with ?

krismanuelp, 2020-05-19:
Doesn't matter. I've found the solution in https://github.com/datitran/raccoon_dataset/issues/45#issuecomment-371041130

closed 2020-05-19 by krismanuelp