google/aiyprojects-raspbian #686: AIY Voice Kit V1 - Is it possible start in local commands only?

grayerbeard, 2020-05-14:


grayerbeard, 2020-05-14:
My V1 been in cupboard a while but now got working although getting it to work complicated by issues with finding right versions of software and instructions.  (Will be posting in [my repository](https://github.com/grayerbeard/aiy) once fully sorted).
My issue now is that its hard to get "Local commands" to be reliable when using (say) "OK Google House Temperatures" to speak temperature readings fed in by MQTT from other R Pis.
Often its mis-heard and instead I get a long spoken answer to the mis interpreted question, in this case often the local weather.

So

Could I start in Local Commands only mode and only switch in to the full "Assistant mode" if I ask for it?   Anyone figured out how to do that ?

manoj7410, 2020-05-14:
@grayerbeard Are you talking about the "local command" demo given at : https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/src/examples/voice/assistant_library_with_local_commands_demo.py#L51 ?

grayerbeard, 2020-05-24:
Yes could be would have to deviise a way to switch modes using the two different demos as a basis.

I will have a go at that.  Thanks.

What I would love to manage is to switch between a mode where it only reacts to my limited list of commands and a mode where it will behave like a normal Google Assistant. i.e. at start you might say "OK Google Special Only" then it would only react to a limited list of my own commands until I say "OK Google Back to normal" when it would revert telling me tomorrows weather etc.