google/aiyprojects-raspbian #688: assistant_library_demo.py for Voice 1.0 not working

mgl1206, 2020-05-24:


manoj7410, 2020-06-11:
Can you try to implement the solution given at https://github.com/google/aiyprojects-raspbian/issues/658#issuecomment-568777735 and see if it works for you ?

jrgleason, 2020-06-16:
@manoj7410 Same issue here. Once you do that you get...

```
~/Code/aiyprojects-raspbian/src/examples/voice $ ./assistant_library_demo.py
Segmentation fault
```

jrgleason, 2020-06-16:
More info (I think)

```
python3.7 -q -X faulthandler ./assistant_library_demo.py
Fatal Python error: Segmentation fault

Current thread 0x76f82ad0 (most recent call first):
  File "/usr/local/lib/python3.7/dist-packages-linux_armv7l/google/assistant/library/assistant.py", line 98 in __init__
  File "/home/pi/AIY-projects-python/src/aiy/assistant/library.py", line 78 in __init__
  File "./assistant_library_demo.py", line 65 in main
  File "./assistant_library_demo.py", line 71 in <module>
Segmentation fault
```

jrgleason, 2020-06-16:
`device_model_id.encode('ASCII')` seems to be the line

jrgleason, 2020-06-16:
Looks like this might be related to https://github.com/googlesamples/assistant-sdk-python/issues/345

manoj7410, 2020-06-16:
Can you run "pip3 install google-assistant-library==1.0.0" and then try the assistant library demo ?

mgl1206, 2020-06-23:
I think I have fixed the issue. I tried out the solution for Issue #658. I opened the file and changed the code from “-1” to “preferred=true”. And so I tried it again but it didn’t work saying that true was undefined. After trying out some other thing so decided to try and follow the path that it was giving me for the original problem and changed “preferred=true” to “-1”. I ran the command “./assistant_library_demo.py” again but instead of the error code it gave me the link for authorization.

rfjohnso, 2020-06-24:
I've just tried the above solution and it is still giving a segmentation fault. RaspPi3b+, version 1 of AIY, latest Raspbian image.
I was able to get around the first error in issue #658 but still getting segmentation fault after pip3 install google-assistant-library.
I tried also changing line 75 with -1, Preferred=True and without either. Did I miss something else?

mgl1206, 2020-06-25:
@rfjohnso ok thx I’ll update I’m comment.

closed 2020-06-25 by mgl1206