google/aiyprojects-raspbian #690: Pi 4 - Voice Bonnet Found - No Sound

KonsultInternetConsulting, 2020-05-27:
Installed on a new Pi4 Raspian manually using HACKING File
Everything installed
Sound Card found
Button works when using 
-----
So sound when running sound test
App Connect button doesnt want to work
Tests fail when trying to check_audio.py - No Sound

Where would I start?

Its like "The lights are on but no body is home!"

manoj7410, 2020-05-27:
Can you run : "arecord -d 5 -f cd test.wav" (And speak for 5 seconds) Then run : "aplay test.wav" (Do you hear what you spoke ?)
Try same commands with earphone connected to the 3.5 mm audio jack.

KonsultInternetConsulting, 2020-05-28:
> Can you run : "arecord -d 5 -f cd test.wav" (And speak for 5 seconds) Then run : "aplay test.wav" (Do you hear what you spoke ?)
> Try same commands with earphone connected to the 3.5 mm audio jack.

I tried these commands and they both seem to run okay, but no sound on soundcard or headphones.

pi@K9OS:~ $ arecord -l
**** List of CAPTURE Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
pi@K9OS:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpigooglevoi [snd_rpi_googlevoicehat_soundcar], device 0: Google voiceHAT SoundCard HiFi voicehat-codec-0 [Google voiceHAT SoundCard HiFi voicehat-codec-0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

manoj7410, 2020-06-11:
That looks fine. Most probably the speaker isn't working. Can you also share the output of "dmesg" as well ?

cjscheller, 2020-06-15:
@manoj7410 I am having same issue as above - running Pi 3B+, voiceHAT soundcard is identified and listed for both arecord and aplay (my output is the same as above), but I'm not able to generate any sound through 2 speakers I have tried or headphones as well as not able to record anything with mic. 

dmesg output includes the following related to audio (can provide more if needed):

> [    4.892292] voicehat-codec voicehat-codec: property 'voicehat_sdmode_delay' not found default 5 mS
...
> [    5.013241] snd-rpi-simple soc:sound: voicehat-codec <-> 3f203000.i2s mapping ok
...
> [   17.213997] voicehat-codec voicehat-codec: Enabling audio amp...
> [   22.365382] voicehat-codec voicehat-codec: Disabling audio amp...
> [   23.041679] voicehat-codec voicehat-codec: Enabling audio amp...
> [   23.057391] voicehat-codec voicehat-codec: Enabling audio amp...
> [   28.042846] voicehat-codec voicehat-codec: Disabling audio amp...
> [  157.590855] voicehat-codec voicehat-codec: Enabling audio amp...
> [  176.245306] voicehat-codec voicehat-codec: Disabling audio amp...
> [  277.887213] voicehat-codec voicehat-codec: Enabling audio amp...
> [  292.398893] voicehat-codec voicehat-codec: Disabling audio amp...
> [  294.798852] voicehat-codec voicehat-codec: Enabling audio amp...
> [  311.562219] voicehat-codec voicehat-codec: Disabling audio amp...

@KonsultInternetConsulting were you able to resolve the issue?

Thanks

KonsultInternetConsulting, 2020-06-15:
Unfortunately not.

I purchased a respeaker hat instead and that works a treat.

I had thought my board was dead or it is incompatable with pi4.

Daren

cjscheller, 2020-06-15:
FWIW, I was able to get it working by following this shell script (linked below) which is specifically for the [Voice Bonnet](https://pinout.xyz/pinout/aiy_voice_bonnet) from AIY voice kit v2 that I was attempting to install:

https://github.com/chiisaa/picroft-google-aiy2-voicebonnet-skill/blob/master/install_AIY2.sh

Originally found in this [Mycroft thread](https://community.mycroft.ai/t/aiy-voice-v2-bonnet-with-a-pi-3/8476/6). I skipped the installation steps related to Mycroft w/o issue.

tanty, 2020-10-19:
> ## Installed on a new Pi4 Raspian manually using HACKING File
> Everything installed
> Sound Card found
> Button works when using
> No sound when running sound test
> App Connect button doesnt want to work
> Tests fail when trying to check_audio.py - No Sound
> 
> Where would I start?
> 
> Its like "The lights are on but no body is home!"

Check https://github.com/google/aiyprojects-raspbian/issues/697#issuecomment-712397021

dmitriykovalev, 2020-11-23:
Please try the new [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120), Voice Bonnet should work on Pi 4.

closed 2020-11-23 by dmitriykovalev