google/aiyprojects-raspbian #692: AIY VOICE KIT - Assistant Demo does not work (assistant_grpc_demo)

cckeung123, 2020-06-13:
Hi,
I am getting an error code when I am trying to run the assistant demo (assistant_grpc_demo).

pi@raspberrypi:~ $ ~/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py

DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
DEBUG:urllib3.connectionpool:https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
INFO:aiy.assistant.grpc:device_model_id: voice-assistant-279901-voice-kit
INFO:aiy.assistant.grpc:device_id: voice-assistant-279901-voice-kit-3d2df07c-eea4-483b-be5b-6bf659af8ac1
DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
DEBUG:urllib3.connectionpool:https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 200 None
INFO:aiy.assistant.grpc:Connecting to embeddedassistant.googleapis.com
/opt/aiy/projects-python/src/aiy/board.py:218: RuntimeWarning: This channel is already in use, continuing anyway.  Use GPIO.setwarnings(False) to disable warnings.
  GPIO.setup(channel, GPIO.OUT)
INFO:root:Press button to start conversation...

INFO:root:Conversation started!
INFO:aiy.assistant.grpc:Recording started.
aplay: playback:2787: read error
INFO:aiy.assistant.grpc:Recording stopped.
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 54, in main
    assistant.conversation()
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 190, in conversation
    keep_talking = self._assist(recorder, wrapped_play, deadline)
  File "/opt/aiy/projects-python/src/aiy/assistant/grpc.py", line 135, in _assist
    for response in self._assistant.Assist(self._requests(recorder), deadline):
  File "/usr/local/lib/python3.5/dist-packages/grpc/_channel.py", line 363, in __next__
    return self._next()
  File "/usr/local/lib/python3.5/dist-packages/grpc/_channel.py", line 357, in _next
    raise self
grpc._channel._Rendezvous: <_Rendezvous of RPC that terminated with:
status = StatusCode.INVALID_ARGUMENT
details = "Invalid 'dialog_state_in': unsupported language_code."
debug_error_string = "{"created":"@1592060445.981045201","description":"Error received from peer","file":"src/core/lib/surface/call.cc","file_line":1099,"grpc_message":"Invalid 'dialog_state_in': unsupported language_code.","grpc_status":3}"
>

Any help much appreciated.
Thanks.

manoj7410, 2020-06-15:
@cckeung123 I see this error : "unsupported language_code". Did you try to change the language of the Assistant ? 
From which country are you trying to run grpc demo ?

cckeung123, 2020-06-15:
@manoj7410  
I come from Hong Kong.
How do I change the language of the Assistant? What language should I change?
Thanks.

manoj7410, 2020-06-16:
Google Assistant SDK supports languages listed at : https://developers.google.com/assistant/sdk/reference/rpc/languages
I have seen this issue in the past form other users from Hong Kong. A user just changed his internet connection and his Assistant language got changed automatically. You should be able to change the language at : https://github.com/google/aiyprojects-raspbian/blob/bc3a748140b391bec562d73c0d518dcf60840859/src/aiy/assistant/grpc.py#L257