google/aiyprojects-raspbian #694: ModuleNotFoundError: No module name 'aiy'

aanthony11, 2020-06-17:
I have a Raspberry Pi 4, and am trying to implement Google Assistant into my project. I am able to run the google assistant when running a googlesamples-assistant-pushtotalk test, but when trying to run the assistant_grpc_demo.py file I run into the "No module named 'aiy' error. I am running the python file in a venv but am still getting the same error. 

[image]

manoj7410, 2020-06-17:
Did you try to run it without VENV as well ? Does that work ?

aanthony11, 2020-06-19:
No running it without VENV doesn't work either

dmitriykovalev, 2020-11-20:
You need to have `aiy` package in `PYTHONPATH` (that's true by default outside of venv). One way to run the example:

```bash
PYTHONPATH=~/AIY-voice-kit-python/src python3 assistant_grpc_demo.py
```

closed 2020-11-20 by dmitriykovalev