google/aiyprojects-raspbian #696: AIY voice kit V1 not working while running the scripts getting error

trsbalaji, 2020-06-22:
Hi,

Followed  google cloud console steps : https://github.com/google/aiyprojects-raspbian/issues/616

Download credentials and tried with python3 same issue seeing.

Could you please help me how to fix this issue?

TypeError: register() takes from 2 to 3 positional arguments but 4 were given
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ python3 assistant_grpc_demo.py 
Traceback (most recent call last):
  File "assistant_grpc_demo.py", line 57, in <module>
    main()
  File "assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
    super().__init__(language_code, volume_percentage)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ls
assistant_grpc_demo.py  assistant_grpc_demo.service  assistant_library_demo.py  assistant_library_with_button_demo.py  assistant_library_with_local_commands_demo.py  cloudspeech_demo.py  recording.wav  voice_recorder.py
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ./assistant_library_with_local_commands_demo.py 
Traceback (most recent call last):
  File "./assistant_library_with_local_commands_demo.py", line 90, in <module>
    main()
  File "./assistant_library_with_local_commands_demo.py", line 83, in main
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
    credentials = _credentials_flow_interactive(client_secrets)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
    webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
TypeError: register() takes from 2 to 3 positional arguments but 4 were given

trsbalaji, 2020-06-22:
Hi,

Another  below error am getting

pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ cd ..
pi@raspberrypi:~/AIY-projects-python/src/examples $ cd ..
pi@raspberrypi:~/AIY-projects-python/src $ cd ..f
bash: cd: ..f: No such file or directory
pi@raspberrypi:~/AIY-projects-python/src $ cd ..
pi@raspberrypi:~/AIY-projects-python $ 
pi@raspberrypi:~/AIY-projects-python $ cd \
> 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ source env/bin/activate
(env) pi@raspberrypi:~ $ cd AIY-voice-kit-python
(env) pi@raspberrypi:~/AIY-voice-kit-python $ 
(env) pi@raspberrypi:~/AIY-voice-kit-python $ cd src/
(env) pi@raspberrypi:~/AIY-voice-kit-python/src $ cd examples/voice
(env) pi@raspberrypi:~/AIY-voice-kit-python/src/examples/voice $ python3 assistant_library_demo.py 
Traceback (most recent call last):
  File "assistant_library_demo.py", line 35, in <module>
    from aiy.assistant import auth_helpers
ModuleNotFoundError: No module named 'aiy'
(env) pi@raspberrypi:~/AIY-voice-kit-python/src/examples/voice $ 
(env) pi@raspberrypi:~/AIY-voice-kit-python/src/examples/voice $ 
(env) pi@raspberrypi:~/AIY-voice-kit-python/src/examples/voice $ python3 assistant_library_demo.py 
Traceback (most recent call last):
  File "assistant_library_demo.py", line 35, in <module>
    from aiy.assistant import auth_helpers
ModuleNotFoundError: No module named 'aiy'

manoj7410, 2020-06-22:
@trsbalaji Did you try to execute assistant_grpc_demo.py without virtual environment?
Have you tried with the image : https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 ?

trsbalaji, 2020-06-22:
Yes, I tried with latest image as well as without virtual environment, same issue seeing

manoj7410, 2020-06-22:
> Hi,
> 
> Followed google cloud console steps : #616
> 
> Download credentials and tried with python3 same issue seeing.
> 
> Could you please help me how to fix this issue?
> 
> TypeError: register() takes from 2 to 3 positional arguments but 4 were given
> pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ python3 assistant_grpc_demo.py Traceback (most recent call last): File "assistant_grpc_demo.py", line 57, in main() File "assistant_grpc_demo.py", line 49, in main language_code=args.language) File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in **init** super().**init**(language_code, volume_percentage) File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in **init** credentials = auth_helpers.get_assistant_credentials() File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials return _try_to_get_credentials(credentials_file) File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials credentials = _credentials_flow_interactive(client_secrets) File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1) TypeError: register() takes from 2 to 3 positional arguments but 4 were given pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ls
> assistant_grpc_demo.py assistant_grpc_demo.service assistant_library_demo.py assistant_library_with_button_demo.py assistant_library_with_local_commands_demo.py cloudspeech_demo.py recording.wav voice_recorder.py
> pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ./assistant_library_with_local_commands_demo.py
> Traceback (most recent call last):
> File "./assistant_library_with_local_commands_demo.py", line 90, in
> main()
> File "./assistant_library_with_local_commands_demo.py", line 83, in main
> credentials = auth_helpers.get_assistant_credentials()
> File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
> return _try_to_get_credentials(credentials_file)
> File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 123, in _try_to_get_credentials
> credentials = _credentials_flow_interactive(client_secrets)
> File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 75, in _credentials_flow_interactive
> webbrowser.register('chromium-browser', None, webbrowser.Chrome('chromium-browser'), -1)
> TypeError: register() takes from 2 to 3 positional arguments but 4 were given

Did you try to implement the solution given at : https://github.com/google/aiyprojects-raspbian/issues/658#issuecomment-568777735 ?

windy54, 2020-07-22:
So I have been having issues and contacted the help desk.
They pointed me at 
_I believe that you are getting an error due to our documentation changes. Our team is working on to fix this issue. We have mentioned the updated instructions at : https://github.com/google/aiyprojects-raspbian/issues/616#issuecomment-643189350. Please follow these instructions to create the credentials.__

I followed these instructions up until step 18 , download credentials. I did not have a download option but could copy the two keys. 
Not sure what file to copy them to, cloud-secrets.json and credentials.json?
what format because all I have copied is plain text?

when i run assistant_gprc.demo.py I still get the same error
"client secretsmust be for a web or installed app"
_