google/aiyprojects-raspbian #697: Cannot install aiy-voicebonnet-soundcard-dkms on latest Raspbian

keyz182, 2020-07-20:
Following instructions [here](https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md), `aiy-voicebonnet-soundcard-dkms` fails to install.

System is up to date.

```bash
$ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
```

```bash
$ sudo apt autoremove
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@ledpi:~ $ sudo apt-get install -y aiy-voicebonnet-soundcard-dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  aiy-voicebonnet-soundcard-dkms
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/32.7 kB of archives.
After this operation, 240 kB of additional disk space will be used.
Selecting previously unselected package aiy-voicebonnet-soundcard-dkms.
(Reading database ... 207200 files and directories currently installed.)
Preparing to unpack .../aiy-voicebonnet-soundcard-dkms_2.0-1_all.deb ...
Unpacking aiy-voicebonnet-soundcard-dkms (2.0-1) ...
Setting up aiy-voicebonnet-soundcard-dkms (2.0-1) ...
Loading new aiy-voicebonnet-soundcard-2.0 DKMS files...
It is likely that 5.4.51-v7+ belongs to a chroot's host
Building for 5.4.51+, 5.4.51-v7+, 5.4.51-v7l+ and 5.4.51-v8+
Building initial module for 5.4.51+
Error! Bad return status for module build on kernel: 5.4.51+ (armv7l)
Consult /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/make.log for more information.
dpkg: error processing package aiy-voicebonnet-soundcard-dkms (--configure):
 installed aiy-voicebonnet-soundcard-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 aiy-voicebonnet-soundcard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

```bash
$ cat /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/make.log
DKMS make.log for aiy-voicebonnet-soundcard-2.0 for kernel 5.4.51+ (armv7l)
Mon 20 Jul 12:36:08 BST 2020
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  AR      /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/built-in.a
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rl6231.o
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:4: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_dai_name’; did you mean ‘stream_name’?
   .codec_dai_name = "rt5645-aif1",
    ^~~~~~~~~~~~~~
    stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:21: error: initialization of ‘struct snd_soc_dai_link_component *’ from incompatible pointer type ‘char *’ [-Werror=incompatible-pointer-types]
   .codec_dai_name = "rt5645-aif1",
                     ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:21: note: (near initialization for ‘snd_rpi_aiy_voicebonnet_dai[0].cpus’)
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_probe’:
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:161:8: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_name’; did you mean ‘stream_name’?
   dai->codec_name = NULL;
        ^~~~~~~~~~
        stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:162:6: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_of_node’
   dai->codec_of_node = of_parse_phandle(dev->of_node,
      ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:164:11: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_of_node’
   if (!dai->codec_of_node) {
           ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:171:9: error: ‘struct snd_soc_dai_link’ has no member named ‘cpu_dai_name’; did you mean ‘stream_name’?
    dai->cpu_dai_name = NULL;
         ^~~~~~~~~~~~
         stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:172:7: error: ‘struct snd_soc_dai_link’ has no member named ‘cpu_of_node’
    dai->cpu_of_node = i2s_node;
       ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:173:9: error: ‘struct snd_soc_dai_link’ has no member named ‘platform_name’; did you mean ‘platforms’?
    dai->platform_name = NULL;
         ^~~~~~~~~~~~~
         platforms
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:174:9: error: ‘struct snd_soc_dai_link’ has no member named ‘platform_of_node’; did you mean ‘platforms’?
    dai->platform_of_node = i2s_node;
         ^~~~~~~~~~~~~~~~
         platforms
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:265: /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3577:3: error: ‘const struct regmap_config’ has no member named ‘use_single_rw’; did you mean ‘use_single_read’?
  .use_single_rw = true,
   ^~~~~~~~~~~~~
   use_single_read
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3577:19: warning: initialization of ‘bool (*)(struct device *, unsigned int)’ {aka ‘_Bool (*)(struct device *, unsigned int)’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  .use_single_rw = true,
                   ^~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3577:19: note: (near initialization for ‘rt5645_regmap.writeable_reg’)
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3593:3: error: ‘const struct regmap_config’ has no member named ‘use_single_rw’; did you mean ‘use_single_read’?
  .use_single_rw = true,
   ^~~~~~~~~~~~~
   use_single_read
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3593:19: warning: initialization of ‘bool (*)(struct device *, unsigned int)’ {aka ‘_Bool (*)(struct device *, unsigned int)’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  .use_single_rw = true,
                   ^~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3593:19: note: (near initialization for ‘rt5650_regmap.writeable_reg’)
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3610:3: error: ‘const struct regmap_config’ has no member named ‘use_single_rw’; did you mean ‘use_single_read’?
  .use_single_rw = true,
   ^~~~~~~~~~~~~
   use_single_read
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3610:19: warning: initialization of ‘bool (*)(struct device *, unsigned int)’ {aka ‘_Bool (*)(struct device *, unsigned int)’} from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
  .use_single_rw = true,
                   ^~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.c:3610:19: note: (near initialization for ‘temp_regmap.writeable_reg’)
make[1]: *** [scripts/Makefile.build:265: /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/rt5645.o] Error 1
make: *** [Makefile:1709: /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
```

keyz182, 2020-07-20:
It appears that `sound/soc.h` changed in kernel 5.3 to change those names.

https://github.com/torvalds/linux/blob/0ecfebd2b52404ae0c54a878c872bb93363ada36/include/sound/soc.h#L930
Vs
https://github.com/torvalds/linux/blob/4d856f72c10ecb060868ed10ff1b1453943fc6c8/include/sound/soc.h#L926

keyz182, 2020-07-20:
This was a red-herring for me, as I have a voice-hat, not the bonnet - D'oh. Though for those with the bonnet, the issue remains, so I wont close the issue. 

There have also been changes in `rt5645.c`.

I was able to get it to compile with these changes, though, as I have the hat, not the bonnet, I can't test:

```diff
diff ./rt5645.c /home/pi/aiy-voicebonnet-soundcard-dkms_2.0-1_all/usr/src/aiy-voicebonnet-soundcard-2.0/rt5645.c
3577,3578c3577
<       .use_single_read = true,
<       .use_single_write = true,
---
>       .use_single_rw = true,
3594,3595c3593
<         .use_single_read = true,
<         .use_single_write = true,
---
>       .use_single_rw = true,
3612,3613c3610
<         .use_single_read = true,
<         .use_single_write = true,
---
>       .use_single_rw = true,
diff ./snd-aiy-voicebonnet.c /home/pi/aiy-voicebonnet-soundcard-dkms_2.0-1_all/usr/src/aiy-voicebonnet-soundcard-2.0/snd-aiy-voicebonnet.c
101,105d100
< static struct snd_soc_dai_link_component snd_rpi_aiy_voicebonnet_codec = {
<         .name = "rt5645",
<         .dai_name = "rt5645-aif1",
< };
<
110c105
<               .codecs = &snd_rpi_aiy_voicebonnet_codec,
---
>               .codec_dai_name = "rt5645-aif1",
166,167c161,162
<               dai->codecs->name = NULL;
<               dai->codecs->of_node = of_parse_phandle(dev->of_node,
---
>               dai->codec_name = NULL;
>               dai->codec_of_node = of_parse_phandle(dev->of_node,
169c164
<               if (!dai->codecs->of_node) {
---
>               if (!dai->codec_of_node) {
176,179c171,174
<                       dai->cpus->dai_name = NULL;
<                       dai->cpus->of_node = i2s_node;
<                       dai->platforms->name = NULL;
<                       dai->platforms->of_node = i2s_node;
---
>                       dai->cpu_dai_name = NULL;
>                       dai->cpu_of_node = i2s_node;
>                       dai->platform_name = NULL;
>                       dai->platform_of_node = i2s_node;
```

nmilenko, 2020-07-21:
I am having a similar issue however my errors are slightly different. Did you ever resolve the issue with your Voice Bonnet? This has been driving me mad lol

`cat /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/make.log
DKMS make.log for aiy-voicebonnet-soundcard-2.0 for kernel 5.4.51+ (armv6l)
Tue 21 Jul 2020 12:31:33 AM EDT
make: Entering directory '/usr/src/linux-headers-5.4.51+'
  AR      /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/built-in.a
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.o
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:4: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_dai_name’; did you mean ‘stream_name’?
   .codec_dai_name = "rt5645-aif1",
    ^~~~~~~~~~~~~~
    stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:21: error: initialization of ‘struct snd_soc_dai_link_component *’ from incompatible pointer type ‘char *’ [-Werror=incompatible-pointer-types]
   .codec_dai_name = "rt5645-aif1",
                     ^~~~~~~~~~~~~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:105:21: note: (near initialization for ‘snd_rpi_aiy_voicebonnet_dai[0].cpus’)
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_probe’:
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:161:8: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_name’; did you mean ‘stream_name’?
   dai->codec_name = NULL;
        ^~~~~~~~~~
        stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:162:6: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_of_node’
   dai->codec_of_node = of_parse_phandle(dev->of_node,
      ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:164:11: error: ‘struct snd_soc_dai_link’ has no member named ‘codec_of_node’
   if (!dai->codec_of_node) {
           ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:171:9: error: ‘struct snd_soc_dai_link’ has no member named ‘cpu_dai_name’; did you mean ‘stream_name’?
    dai->cpu_dai_name = NULL;
         ^~~~~~~~~~~~
         stream_name
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:172:7: error: ‘struct snd_soc_dai_link’ has no member named ‘cpu_of_node’
    dai->cpu_of_node = i2s_node;
       ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:173:9: error: ‘struct snd_soc_dai_link’ has no member named ‘platform_name’; did you mean ‘platforms’?
    dai->platform_name = NULL;
         ^~~~~~~~~~~~~
         platforms
/var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.c:174:9: error: ‘struct snd_soc_dai_link’ has no member named ‘platform_of_node’; did you mean ‘platforms’?
    dai->platform_of_node = i2s_node;
         ^~~~~~~~~~~~~~~~
         platforms
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:266: /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build/snd-aiy-voicebonnet.o] Error 1
make: *** [Makefile:1709: /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.4.51+'
`

keyz182, 2020-07-21:
@nmilenko those errors are identical to mine after a _quick_ look. The changes in the diff should get it to at least compile.

As I mentioned though - I was barking up the wrong tree - I had a Voice Hat, not Bonnet, so I have no idea if my changes actually work as I can't test them.

Try something like 
```bash
sudo mkdir /usr/src/aiy-voicebonnet-soundcard-2.0fixed/
sudo cp /var/lib/dkms/aiy-voicebonnet-soundcard/2.0/source/* /usr/src/aiy-voicebonnet-soundcard-2.0fixed/
sudo apt remove aiy-voicebonnet-soundcard-dkms
```

Then apply the changes in the diff in https://github.com/google/aiyprojects-raspbian/issues/697#issuecomment-661103336 to the files in `/usr/src/aiy-voicebonnet-soundcard-2.0fixed/`. Then:

```
sudo dkms add -m aiy-voicebonnet-soundcard -v 2.0fixed
sudo dkms build -m aiy-voicebonnet-soundcard -v 2.0fixed
sudo dkms install -m aiy-voicebonnet-soundcard -v 2.0fixed
```

Can't really help you beyond that I'm afraid, as I'm not particularly familiar with the code, and I don't have the hardware to test with.

blcarlson01, 2020-07-26:
Unfortunately I tried your steps and I still received the "Error! Bad return status for module build on kernel: 5.4.51+ (armv7l)"

I have the Voice Hat as well and I'm trying to decide if its a lost cause.  The only other solution I've seen online is use the most recent image and don't update/upgrade.

protoz, 2020-07-29:
Same issue for me on the Voice Hat using the packages in apt.  I haven't tried compiling from source yet.

Squeack, 2020-08-04:
I followed keyz182's instructions and got the modules to at least compile. Note that his diffs are from new version to original, so the arrows seems to point the wrong way. Find the green lines in the code and replace them with the red versions.

The dkms build and install commands then ran smoothly, but following the HACKING.md instructions as far as the verify audio, the Voice hat still did not work, and no audio devices were shown by aplay -L or arecord -L. Adding a line to /boot/config.txt and rebooting made sure the modules were loaded and running properly:
`dtoverlay=googlevoicehat-soundcard`
This problem was solved by following section 4 of HACKING.md, where the python tools are downloaded. There is a check_audio.py script which correctly identified the problem.

thechamp21, 2020-09-14:
@Squeack could you please go into more detail on how you fixed this? I'm a total noob, but basically I've been going into these files and changing the green lines of code mentioned with the red lines of code (editing using nano). No clue if this is the right way of going about this, but would love any help you might be able to provide.

viraniac, 2020-09-29:
I have created a modified deb file to make it easy for others to install this package. The deb file can be downloaded from this [link](https://drive.google.com/file/d/11GYLbXB6zXEFu7K--rhI42m7njV_G_6N/view?usp=sharing).

To install, download the file and run the following commands

```
sudo apt-get install -y dkms raspberrypi-kernel-headers
sudo dpkg -i aiy-voicebonnet-soundcard-dkms_2.0-1_all.deb
```

viraniac, 2020-09-30:
@keyz182 I have both voicehat and voicebonnet. The changes are working for voicehat, but even though the modules get compiled for voicebonnet, it gives null pointer exception when its getting loaded. 

I am not a kernel developer so it might take some time for me to solve it. Still trying to solve it though.

I have pasted the backtrace of the exception [here](http://paste.debian.net/1165246/).

viraniac, 2020-10-03:
After trying tons of different things, finally got it to work on the voicebonnet too. Download the fixed deb file from [here](https://drive.google.com/file/d/1AN8PdOevPZz6yb8Z2AspeOqACdZAL0eA/view?usp=sharing) 

To install, download the file and run the following commands
```
sudo apt-get install -y dkms raspberrypi-kernel-headers
sudo dpkg -i aiy-voicebonnet-soundcard-dkms_2.0-1_all.deb
```

Here is the patch file for the changes that I have done. Mostly they are same as what @keyz182  have done. I have applied few small tweaks to it to get it to work.

```
diff -ruN old/rt5645.c new/rt5645.c
--- old/rt5645.c	2019-11-07 00:36:24.000000000 +0000
+++ new/rt5645.c	2020-09-29 12:22:42.000000000 +0100
@@ -3574,7 +3574,8 @@
 static const struct regmap_config rt5645_regmap = {
 	.reg_bits = 8,
 	.val_bits = 16,
-	.use_single_rw = true,
+	.use_single_read = true,
+	.use_single_write = true,
 	.max_register = RT5645_VENDOR_ID2 + 1 + (ARRAY_SIZE(rt5645_ranges) *
 					       RT5645_PR_SPACING),
 	.volatile_reg = rt5645_volatile_register,
@@ -3590,7 +3591,8 @@
 static const struct regmap_config rt5650_regmap = {
 	.reg_bits = 8,
 	.val_bits = 16,
-	.use_single_rw = true,
+	.use_single_read = true,
+	.use_single_write = true,
 	.max_register = RT5645_VENDOR_ID2 + 1 + (ARRAY_SIZE(rt5645_ranges) *
 					       RT5645_PR_SPACING),
 	.volatile_reg = rt5645_volatile_register,
@@ -3607,7 +3609,8 @@
 	.name="nocache",
 	.reg_bits = 8,
 	.val_bits = 16,
-	.use_single_rw = true,
+	.use_single_read = true,
+	.use_single_write = true,
 	.max_register = RT5645_VENDOR_ID2 + 1,
 	.cache_type = REGCACHE_NONE,
 };
diff -ruN old/snd-aiy-voicebonnet.c new/snd-aiy-voicebonnet.c
--- old/snd-aiy-voicebonnet.c	2019-11-07 00:36:24.000000000 +0000
+++ new/snd-aiy-voicebonnet.c	2020-10-03 11:03:28.428630886 +0100
@@ -98,17 +98,20 @@
 	.hw_params = snd_rpi_aiy_voicebonnet_hw_params,
 };
 
-static struct snd_soc_dai_link snd_rpi_aiy_voicebonnet_dai[] = {
-	{
-		.name = "rt5645",
-		.stream_name = "Google AIY Voice Bonnet SoundCard HiFi",
-		.codec_dai_name = "rt5645-aif1",
-		.dai_fmt = SND_SOC_DAIFMT_I2S |
-			   SND_SOC_DAIFMT_NB_NF |
-			   SND_SOC_DAIFMT_CBS_CFS,
-		.ops = &snd_rpi_aiy_voicebonnet_ops,
-		.init = snd_rpi_aiy_voicebonnet_init
-	},
+SND_SOC_DAILINK_DEFS(pcm,
+	DAILINK_COMP_ARRAY(COMP_EMPTY()),
+	DAILINK_COMP_ARRAY(COMP_CODEC(NULL, "rt5645-aif1")),
+	DAILINK_COMP_ARRAY(COMP_EMPTY()));
+
+static struct snd_soc_dai_link snd_rpi_aiy_voicebonnet_dai = {
+	.name = "rt5645",
+	.stream_name = "Google AIY Voice Bonnet SoundCard HiFi",
+	.init = snd_rpi_aiy_voicebonnet_init,
+	.ops = &snd_rpi_aiy_voicebonnet_ops,
+	.dai_fmt = SND_SOC_DAIFMT_I2S |
+		   SND_SOC_DAIFMT_NB_NF |
+		   SND_SOC_DAIFMT_CBS_CFS,
+	SND_SOC_DAILINK_REG(pcm),
 };
 
 static const struct snd_soc_dapm_widget voicebonnet_widgets[] = {
@@ -137,8 +140,8 @@
 static struct snd_soc_card snd_rpi_aiy_voicebonnet = {
 	.name = "snd_rpi_aiy_voicebonnet",
 	.owner = THIS_MODULE,
-	.dai_link = snd_rpi_aiy_voicebonnet_dai,
-	.num_links = ARRAY_SIZE(snd_rpi_aiy_voicebonnet_dai),
+	.dai_link = &snd_rpi_aiy_voicebonnet_dai,
+	.num_links = 1,
 	.dapm_routes = voicebonnet_audio_map,
 	.num_dapm_routes = ARRAY_SIZE(voicebonnet_audio_map),
 	.dapm_widgets = voicebonnet_widgets,
@@ -150,7 +153,7 @@
 
 static int snd_rpi_aiy_voicebonnet_probe(struct platform_device *pdev) {
 	int ret = 0;
-	struct snd_soc_dai_link *dai = &snd_rpi_aiy_voicebonnet_dai[0];
+	struct snd_soc_dai_link *dai = &snd_rpi_aiy_voicebonnet_dai;
 	struct snd_soc_card *card = &snd_rpi_aiy_voicebonnet;
 	struct device *dev = &pdev->dev;
 	struct device_node *i2s_node;
@@ -158,20 +161,17 @@
 	card->dev = dev;
 
 	if (dev->of_node) {
-		dai->codec_name = NULL;
-		dai->codec_of_node = of_parse_phandle(dev->of_node,
+		dai->codecs->of_node = of_parse_phandle(dev->of_node,
 			"aiy-voicebonnet,audio-codec", 0);
-		if (!dai->codec_of_node) {
+		if (!dai->codecs->of_node) {
 			dev_err(dev, "can't parse codec node\n");
 			return -EINVAL;
 		}
 
 		i2s_node = of_parse_phandle(dev->of_node, "i2s-controller", 0);
 		if (i2s_node) {
-			dai->cpu_dai_name = NULL;
-			dai->cpu_of_node = i2s_node;
-			dai->platform_name = NULL;
-			dai->platform_of_node = i2s_node;
+			dai->cpus->of_node = i2s_node;
+			dai->platforms->of_node = i2s_node;
 		}
 	}
 
```

gogus, 2020-10-03:
I tried your solution @viraniac unfortunately no luck, still, I cannot find the card by aplay -l.

viraniac, 2020-10-04:
> I tried your solution @viraniac unfortunately no luck, still, I cannot find the card by aplay -l.

@gogus I tried my deb file thrice before posting it here, so I know for sure it works and unfortunately your post didnt had enough information for me to deduce what went wrong on your device. Could you please give me more information to deduce why it didnt worked for you? Following points will be helpful.

- Were you installing my deb file on top of a fresh installation of raspbian?
- Did you had the older of version of aiy-voicebonnet-soundcard-dkmsinstalled? If yes, I will suggest you uninstall it first using "sudo apt-get remove aiy-voicebonnet-soundcard-dkms" and then reboot the system before installing my deb file.
- If you were doing it on a fresh raspbian installation, did you followed the rest of instructions from Hacking.md to install and configure PulseAudio?
- Can you share the output that you got after running the commands that I have given in my previous post?
- Could you please share the output of dmesg command?
- Any other information will be nice too.

gogus, 2020-10-04:
@viraniac 

Thanks for your answer and help. Answers for your questions:

1) Voice HAT 
2) On fresh latest raspbian installation 
3) I removed old version.
4) I followed Hacking.md for the everything and at the end I just uninstalled aiy-voicebonnet-soundcard-dkms.
5,6) https://gist.github.com/gogus/e8d2d6fdd08111b64230e47c9fa3727e
7) Raspberry PI 3, on the ready-image everything was working properly.

viraniac, 2020-10-04:
@gogus 

Thanks for sharing those outputs. Your dmesg output shows that the system never tried to load any of the aiy card drivers.

As you have Voice Hat, could you please confirm if you have `dtoverlay=googlevoicehat-soundcard` in your `/boot/config.txt` file. if its there, can you share the output of `dtoverlay -l` command?

If the entry is not there in your /boot/config.txt file, you have to manually load the dtoverlay on every reboot using 
`sudo dtoverlay googlevoicehat-soundcard`

gogus, 2020-10-04:
@viraniac Thanks, the entry wasn't existing there. I added that and it seems it's working. Thank you very much for your help!

viraniac, 2020-10-04:
I have noticed that sometimes on voicebonnet, the rt5645 module was not able to identify the device, hence I have replaced the code for rt5645.* and rl6231.* with their corresponding code from the official raspberrypi kernel.

Also I have hosted my modifications on github now and have bumped up the deb file's version number to denote it is a nmu release. I am going to update the link to the file above in my previous post.

gogus, 2020-10-04:
@viraniac I just have another issue right now that if I record the sound from mic using arecord it's only recording for the first 5 seconds. When I will say something after this first 5 seconds and then I will use aplay I cannot hear anything. Do you know maybe what's can be the issue of that?

viraniac, 2020-10-04:
> @viraniac I just have another issue right now that if I record the sound from mic using arecord it's only recording for the first 5 seconds. When I will say something after this first 5 seconds and then I will use aplay I cannot hear anything. Do you know maybe what's can be the issue of that?

No idea, let me try that out.

BTW I just found out, voice hat doesnt really uses that aiy-voicebonnet-soundcard-dkms drivers. All it requires to work is that dtoverlay line. I have raised a issue so that google can update their hacking.md file or probably I will raise a pull request for that too :)

viraniac, 2020-10-04:
@gogus I have no idea why its doing that. Mine is only recording first 2 seconds of audio that too in such a low volume that I can barely hear what it has recorded. I am not sure what is the issue here. I will probably check that next weekend if I get time. No promises though

thechamp21, 2020-10-06:
@viraniac thanks for your help on this! I followed your instructions and it worked perfectly. Can't wait to finally use my voice kit!

viraniac, 2020-10-07:
> @viraniac I just have another issue right now that if I record the sound from mic using arecord it's only recording for the first 5 seconds. When I will say something after this first 5 seconds and then I will use aplay I cannot hear anything. Do you know maybe what's can be the issue of that?

@gogus I figured out my low volume issue and I believe the solution will help you too with that 5 second recording issue

Try creating `~/.asoundrc` file with the following content

```
pcm.softvol {
    type softvol
    slave.pcm dmix
    control {
        name Master
        card sndrpigooglevoi
    }
}

pcm.micboost {
    type route
    slave.pcm dsnoop
    ttable {
        0.0 30.0
        1.1 30.0
    }
}

pcm.!default {
    type asym
    playback.pcm "plug:softvol"
    capture.pcm "plug:micboost"
}

ctl.!default {
    type hw
    card sndrpigooglevoi
}

defaults.pcm.dmix.!card sndrpigooglevoi
defaults.pcm.dsnoop.!card sndrpigooglevoi
```

Also try adding these entries in your `/boot/config.txt` file
```
dtparam=i2s=on
dtoverlay=i2s-mmap
```

bravozulu1979, 2020-10-10:
@viraniac  You are my hero! It works ... have been waiting for this for 3 months now and I can't believe Google did not fix such a basic capability of the bonnet. Thank you!!

tanty, 2020-10-19:
> I have noticed that sometimes on voicebonnet, the rt5645 module was not able to identify the device, hence I have replaced the code for rt5645.* and rl6231.* with their corresponding code from the official raspberrypi kernel hoping that will resolve the issue. So far I didn't got the issue again even though I cold booted and rebooted my voice kit v2 multiple times, but if anyone faces that issue do let me know. Also remember that the issue goes away by doing a reboot, so its nothing really that critical.
> 
> Also I have hosted my modifications on github now and have bumped up the deb file's version number to denote it is a nmu release. I am going to update the link to the file above in my previous post.

First, thanks a lot for your work @viraniac .

I have a RPi 4 with a AIY Voice Kit v2 (Voice Bonnet).

I'm able to install your patched [aiy-voicebonnet-soundcard-dkms_2.0-1+nmu1_all.deb](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/2.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_2.0-1+nmu1_all.deb) package in the latest Raspbian Buster Desktop image.

However, on booting, I get the following:

```
$ dmesg | grep rt5645
[    4.343349] rt5645 1-001a: 1-001a supply avdd not found, using dummy regulator
[    4.343463] rt5645 1-001a: 1-001a supply cpvdd not found, using dummy regulator
[    4.781845] rt5645 1-001a: Device with ID register 0xd6d0dc04 is not rt5645 or rt5650
[    4.790082] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    4.807062] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    5.287268] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    5.300197] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    5.356977] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    5.369455] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
[    5.397795] snd-soc-aiy-voicebonnet soc:sound: ASoC: failed to init link rt5645: -517
```

LEDs and button work but the speaker and mic won't work.

If I unload and load again the modules, it starts working:

```
sudo rmmod snd_aiy_voicebonnet
sudo rmmod rt5645
sudo modprobe snd_aiy_voicebonnet
$ dmesg | grep rt5645
...
[  562.666986] rt5645 1-001a: 1-001a supply avdd not found, using dummy regulator
[  562.667105] rt5645 1-001a: 1-001a supply cpvdd not found, using dummy regulator
[  563.119844] snd-soc-aiy-voicebonnet soc:sound: rt5645-aif1 <-> fe203000.i2s mapping o
```

Any idea?

viraniac, 2020-10-20:
@tanty This was the exact same issue I was talking about where I said that sometimes rt5645 was not able to identify the device. But for me that was gone since I pulled the updated files from upstream kernel sources.

Can you try removing and reinstalling the aiy-voicebonnet-soundcard-dkms package?

Can you share some information about your setup? Your kernel version? Also what power supply are you using, is it the official raspberrypi power supply for pi4? I am not sure if that is the issue, but maybe the pi might not be getting enough power?

Also can you try updating all the packages and pi4 firmware?

LoganTheRed, 2020-10-30:
> After trying tons of different things, finally got it to work on the voicebonnet too. Download the fixed deb file from [here](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/2.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_2.0-1+nmu1_all.deb).

I can confirm this resolved this same issue for me on raspbian kernel 5.4.72

dmitriykovalev, 2020-11-23:
Driver is fixed now, please try the new [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120) or install `aiy-voicebonnet-soundcard-dkms` (current version is `3.0-1`).

tanty, 2020-11-23:
Thanks a lot, @dmitriykovalev .

However, with this new version, I'm still experiencing what's described at https://github.com/google/aiyprojects-raspbian/issues/697#issuecomment-712397021

dmitriykovalev, 2020-11-23:
@tanty I did test on Pi 4 and didn't notice any issues. Can you try our [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120) instead of package installation? You can also try to run `sudo apt update && sudo apt upgrade` on our image to upgrade to `5.4.72` kernel as @LoganTheRed suggested.

aandroide, 2021-01-10:
Has the problem been solved with the kernel released on January 8th?

xamindar, 2021-02-22:
Nope, still a problem. Amazing.

viraniac, 2021-02-26:
@aandroide and @xamindar,

I have created updated drivers that you can get from my [repository](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms). You can install [these](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/3.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_3.0-1+nmu1_all.deb) until google updates their own drivers.

builderjer, 2022-12-08:
I know this is old, but I have 3 voicebonnets and would love to put them to use.
I have tried the solution by @viraniac [here](https://github.com/google/aiyprojects-raspbian/issues/697#issuecomment-786706171) but without success on a new kernel.  I am NOT a kernel guy so don't even know where to start.  Any help would be appreciated.

OS: latest RPiOS lite (bullseye) https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit
Kernel: 5.15.76-v8+
Error log.
```
cat /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/make.log 
DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.15.76-v8+ (aarch64)
Thu  8 Dec 17:21:40 GMT 2022
make: Entering directory '/usr/src/linux-headers-5.15.76-v8+'
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rl6231.o
In file included from /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:25:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.h:12:10: fatal error: sound/rt5645.h: No such file or directory
   12 | #include <sound/rt5645.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.c:32:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.h:12:10: fatal error: sound/rt5645.h: No such file or directory
   12 | #include <sound/rt5645.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o] Error 1
make: *** [Makefile:1900: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.76-v8+'
```