google/aiyprojects-raspbian #699: cloudspeech_demo.py not running but can run assistant_library_demo.py and assistant_grpc_demo.py

pythonnewbi, 2020-08-03:
Hi, I'm new to programming but have gotten the google v1 voice kit partially running on raspberry pi 3b. I can run the assistant_grpc_demo.py and assistant_library_demo.py (I needed to fix a segmentation fault to get it working.) However, no matter what I do I can't get the cloudspeech_demo.py working.  Here is the error:

> 
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ./cloudspeech_demo.py 
INFO:root:Initializing for language en_US...
Traceback (most recent call last):
  File "./cloudspeech_demo.py", line 71, in <module>
    main()
  File "./cloudspeech_demo.py", line 46, in main
    client = CloudSpeechClient()
  File "/home/pi/AIY-projects-python/src/aiy/cloudspeech.py", line 62, in __init__
    credentials = service_account.Credentials.from_service_account_file(service_accout_file)
  File "/usr/local/lib/python3.7/dist-packages/google/oauth2/service_account.py", line 209, in from_service_account_file
    filename, require=['client_email', 'token_uri'])
  File "/usr/local/lib/python3.7/dist-packages/google/auth/_service_account_info.py", line 73, in from_filename
    return data, from_dict(data, require=require)
  File "/usr/local/lib/python3.7/dist-packages/google/auth/_service_account_info.py", line 51, in from_dict
    'fields {}.'.format(', '.join(missing)))
ValueError: Service account info was not in the expected format, missing fields token_uri, client_email.

> ``
Thanks!

manoj7410, 2020-08-03:
@pythonnewbi Can you confirm if you are following the steps provided at : https://aiyprojects.withgoogle.com/voice/#makers-guide--custom-voice-user-interface and have created cloud_speech.json credentials file after enabling the billing ?

pythonnewbi, 2020-08-03:
We just figured it out a short while ago.  The google cloud website changed, but we finally figured out how to turn on the json.  
Thanks!

closed 2020-11-20 by dmitriykovalev