google/aiyprojects-raspbian #701: Vision Bonnet usb type C ?

javixuu, 2020-08-11:
Hi everyone,

**I recently got the AIY Vision Kit 1.1.  After playing with it, I discover that in previous versions there was a Usb Type C soldered directly on the Vision Bonnet, but not in v1.1.**

This picture shows the version with Usb type C that I refered.
[image]

I would like to know the purpose of that usb C. 

Could it be used to connect the bonnet to a Rpi through USB and remove the flex cable? I understand there could be speed limitations but I would like to have more info about this if possible.

In Vision Bonnet from kit 1.1 has the pins for soldering the Usb Type C, so I imagine it could be solder.

Tks AIY team!

dmitriykovalev, 2020-11-23:
Unfortunately USB-C connector is not officially supported, it was originally used for debugging purposes during product development cycle, there is no way to use it now. It cannot replace MIPI interface (flex cable) in any way.

closed 2020-11-23 by dmitriykovalev

javixuu, 2020-11-27:
> Unfortunately USB-C connector is not officially supported, it was originally used for debugging purposes during product development cycle, there is no way to use it now. It cannot replace MIPI interface (flex cable) in any way.

tks for the info, I will switch to coral in that case