google/aiyprojects-raspbian #702: Distance Sensor- GPIO expansion pins not working for input

KeiserSouze, 2020-08-13:
I'm trying to use an HC-SR04 distance sensor with the AIY voice kit. I used this setup, except using bonnet PIN_A and PIN_B instead of RPi GPIO23 and 24, and I used a 1000 Ohm and 2000 Ohm resistor rather than a 330 and a 470.

[image]

Here is my script:

```
from gpiozero import DistanceSensor
from time import sleep
from aiy.pins import PIN_A
from aiy.pins import PIN_B

sensor = DistanceSensor(echo=PIN_A, trigger=PIN_B)

while True:
    print('Distance to nearest object is ', sensor.distance, 'm')
    sleep(1)
```
When I run it, I get the following error:

> Traceback (most recent call last):
>   File "sensor_demo.py", line 6, in <module>
>     sensor = DistanceSensor(echo=PIN_A, trigger=PIN_B)
>   File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 124, in __call__
>     self = super(GPIOMeta, cls).__call__(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 856, in __init__
>     partial=partial, ignore=frozenset({None}), pin_factory=pin_factory
>   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 277, in __init__
>     pin_factory=pin_factory)
>   File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 197, in __init__
>     super(EventsMixin, self).__init__(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 103, in __init__
>     self.pin.pull = pull
>   File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 337, in <lambda>
>     lambda self, value: self._set_pull(value),
>   File "/home/pi/AIY-projects-python/src/aiy/pins.py", line 605, in _set_pull
>     'Only pull up is supported right now (%s)' % pull)
> gpiozero.exc.PinFixedPull: Only pull up is supported right now (down)

When I run it on a RPi 1 B+ directly (with RPi GPIO pins), it works fine, so I think it has to do with the GPIO expansion pins on the bonnet, especially when they are used for input, like [here](https://github.com/google/aiyprojects-raspbian/issues/331#issuecomment-405769640), [here](https://stackoverflow.com/questions/61507324/gpio-zero-button-press-error-with-aiy-pins-py), and [here](https://stackoverflow.com/questions/59973849/is-there-any-example-of-how-to-use-the-digital-input-or-analog-input-of-gpio-on).

Is there any way to make this work with the GPIO expansion pins, or do I have to take the RPi 0 out of the box and use its pins?

rudjer, 2020-09-12:
I would be very interested in any light that might be shed on this problem.  I've also had no success with the GPIO pins
on my v2 voice kit, even with extremely simple applications as I've described previously in the second "here" in the post above.

KKeiser, 2020-09-12:
I was able to get the GPIO expansion pins for LEDs and Servo motors, but not for the Distance Sensor. So my best guess is that they work well for output, but not for input.

Eventually, I just separated the bonnet from the Pi 0W and then reconnected them with jumper wires according to [this pinout](https://github.com/google/aiyprojects-raspbian/issues/331#issuecomment-404990798). That gave me access to the Pi's other pins.

Right now I've got a distance sensor connected to the pi pins and a servo motor connected to the expansion pins, and that has worked just fine.

Hernane007, 2022-09-08:
Hello guys. I have the same problem, when I try to run a code to check the distance through the ultrasonic sensor, the following error appears (gpiozero.exc.PinFixedPull: Only pull up is supported right now (down)). Has anyone had a solution using the pins of expansion ?