google/aiyprojects-raspbian #708: Vision Kit Hangs and/or stops responding when on monitor

khaledallen, 2020-10-06:
I got a vision kit last week and set it up fine. The Joy detection demo works fine. But today was the first day I hooked it up to a monitor and ran Raspbian, at which point it started hanging consistently.

1. Linux app bar doesn't load icons or icons are loaded by are not clickable
2. Raspbian generally responding so slowly it eventually stops responding entirely,
3. SSH terminal stops responding

So, the kit becomes totally unresponsive. In some cases, the Joy Detection demo would continue to run and work behind an unresponsive Raspbian. I also had it freeze when I was running `image_identification_camera.py` in which it would continue to update the image in the monitor, but not the id text, and the keyboard became totally unresponsive so I could not stop the demo and eventually had to unplug it.

In all cases, it maintains an image on the monitor (Raspbian desktop or the camera image), but it stops accepting keyboard/mouse or SSH input.

I'm not really sure what to look for or report, so let me know if there's anything I can add to this report to make it more informative.

manoj7410, 2020-10-08:
@khaledallen Are there any specific steps to reproduce this issue? I just connected my vision kit with the monitor and it is working fine. I am using release from https://github.com/google/aiyprojects-raspbian/releases/tag/v20191113 along with 2.1 amp of power adaptor to boot the kit.

khaledallen, 2020-10-08:
@manoj7410 The only actions I took were to turn it on, and then when Raspbian started, it was slow with broken UI, and the welcome to raspbian window would become unresponsive if I closed it or attempted to complete setup. At one point, the UI was available and I tried to shutdown via the shutdown button, but the dialog became unresponsive when I clicked any of the options or the close button. I was able to open a terminal window and run things in it, but closing it made it unresponsive, and I could not open a new one.

So in general, attempts to open windows in Raspbian were agonizingly slow, and closing said windows made them unresponsive.

In one case, it was working fine (in the sense that it was responding to mouse and terminal keyboard inputs through a terminal window in Raspbian. Still not full UI) and I stepped away for a minute. When I got back, it had stopped responding to any inputs. I had to unplug it.

dmitriykovalev, 2020-11-20:
Please make sure to use 2.1A power adapter. Does Raspberry Pi work fine when Vision Bonnet is disconnected?