google/aiyprojects-raspbian #710: Access PIN_A, PIN_B on VoiceBonnet + RaspberryPi 4

dominickchen, 2020-10-10:
I'm trying to port a working script from an original RaspberryPi WH Zero+VoiceBonnet to a Raspberry Pi4 + VoiceBonnet setup. The RPI4+VoiceBonnet have the speaker and mic working properly, with the latest AIY-VoiceKit image (2019-11) flashed on microSD. The script turns on and off a solenoid. I need to move it to a RPI4 because I need to connect with an USB mic.

But when I try to access the GPIO pins from Python script, it doesn't found the AIY_USER0 and AIY_USER1 pins.

The script halts at the following:
```
from gpiozero import LED
from aiy.pins import PIN_B, PIN_A
sol_A = LED(PIN_A)
sol_B = LED(PIN_B)
```
When executed, the script doesn't exit but freezes, and when I terminate it with Control+C, the traceback shows:
`FileNotFoundError:` [Errno 2] No such file or directory: '/sys/class/gpio/AIY_USER0/active_low'

When I `ls /sys/class/gpio/`,  it only shows `AIY_PA16`. 

I checked on a WH Zero+VoiceBonnet, and `ls /sys/class/gpio/` shows `AIY_USER0` and `AIY_USER1`.

I tried to change in `AIY-projects-python/src/aiy/pins.py` as the following:

```
l108    PIN_A = AIYPinSpec(GpioSpec(PIN_OFFSET, 2, 'AIY_PA16', active_low=False), PwmSpec(0, 'pwm0'))
l109    PIN_B = AIYPinSpec(GpioSpec(PIN_OFFSET, 3, 'AIY_PA17', active_low=False), PwmSpec(1, 'pwm1'))
l110    PIN_C = AIYPinSpec(GpioSpec(PIN_OFFSET, 8, 'AIY_USER2', active_low=False), PwmSpec(2, 'pwm2'))
l111    PIN_D = AIYPinSpec(GpioSpec(PIN_OFFSET, 9, 'AIY_USER3', active_low=False), PwmSpec(3, 'pwm3'))
```

I also tried to manually adding to the GPIO:
`echo $((497 + 2)) > /sys/class/gpio/export`
but it returns `-bash: echo: write error: Device or resource busy`.

Is there any way to access PIN_A and PIN_B from Raspberry Pi 4?

viraniac, 2020-10-18:
I checked on my pi4 and voicebonnet and also saw the same behavior.

I have created a shell script to get the pins to show up. I hope this will be able to help you too. I am meanwhile looking into the kernel module to try to understand whats going wrong with it.

```
#!/bin/bash

base=`ls -al /sys/class/gpio/gpiochip* | grep aiy | grep -Po '[0-9]+$'`
offset=0

while [ $offset -lt 15 ]; do
    sudo sh -c "echo $(($base + $offset)) > /sys/class/gpio/export"
    offset=$(($offset + 1))
done
```

viraniac, 2020-10-18:
Went through the kernel module code. That looks good.

Instead the issue is [here](https://github.com/google/aiyprojects-raspbian/blob/adb1fa483423a12b87e44343c488b2c9e32c338e/src/aiy/pins.py#L93)

The offset is hardcoded. The offset is 497 for pi zero, but it comes as 489 for pi 4. so you should be able to fix the issue by changing the value of PIN_OFFSET in aiy/pins.py file.

dmitriykovalev, 2020-11-23:
Thank you! That's fixed in the latest source code and [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120).

closed 2020-11-23 by dmitriykovalev