google/aiyprojects-raspbian #716: cloudspeech_demo.py fails:  AttributeError: module 'google.cloud.speech' has no attribute 'types'

bbrendon, 2020-11-26:
I tried doing a git pull to get possibly updated but that didn't change anything. Any ideas?

```
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $ ./cloudspeech_demo.py
Traceback (most recent call last):
  File "./cloudspeech_demo.py", line 22, in <module>
    from aiy.cloudspeech import CloudSpeechClient
  File "/home/pi/AIY-projects-python/src/aiy/cloudspeech.py", line 40, in <module>
    END_OF_SINGLE_UTTERANCE = speech.types.StreamingRecognizeResponse.END_OF_SINGLE_UTTERANCE
AttributeError: module 'google.cloud.speech' has no attribute 'types'
pi@raspberrypi:~/AIY-projects-python/src/examples/voice $

```

rabbitmcrabbit, 2020-12-05:
Solved it with the following three changes:

```python
from google.cloud import speech_v1 as speech
...
END_OF_SINGLE_UTTERANCE = speech.types.StreamingRecognizeResponse.SpeechEventType.END_OF_SINGLE_UTTERANCE
...
encoding=speech.types.RecognitionConfig.AudioEncoding.LINEAR16,
...
```

bbrendon, 2020-12-06:
@rabbitmcrabbit  what file did you modify? Do you have a diff?

rabbitmcrabbit, 2020-12-06:
This was an edit to `~/AIY-projects-python/src/aiy/cloudspeech.py`.
My apologies, I'm writing from a machine where it's not straightforward to attach a diff, but the start of each line remains unchanged, so you should be able to find the corresponding places.

bbrendon, 2020-12-21:
Thanks. I got past this. I would have never figured that out.

```
diff --git a/src/aiy/cloudspeech.py b/src/aiy/cloudspeech.py
index d8bcd57..ddd8225 100644
--- a/src/aiy/cloudspeech.py
+++ b/src/aiy/cloudspeech.py
@@ -32,12 +32,15 @@ import os
 import logging

 os.environ['GRPC_POLL_STRATEGY'] = 'epoll1'
-from google.cloud import speech
+#from google.cloud import speech
+from google.cloud import speech_v1 as speech
 from google.oauth2 import service_account

 from aiy.voice.audio import AudioFormat, Recorder

-END_OF_SINGLE_UTTERANCE = speech.types.StreamingRecognizeResponse.END_OF_SINGLE_UTTERANCE
+#END_OF_SINGLE_UTTERANCE = speech.types.StreamingRecognizeResponse.END_OF_SINGLE_UTTERANCE
+END_OF_SINGLE_UTTERANCE = speech.types.StreamingRecognizeResponse.SpeechEventType.END_OF_SINGLE_UTTERANCE
+
 AUDIO_SAMPLE_RATE_HZ = 16000
 AUDIO_FORMAT=AudioFormat(sample_rate_hz=AUDIO_SAMPLE_RATE_HZ,
                          num_channels=1,
@@ -64,7 +67,8 @@ class CloudSpeechClient:

     def _make_config(self, language_code, hint_phrases):
         return speech.types.RecognitionConfig(
-            encoding=speech.types.RecognitionConfig.LINEAR16,
+            #encoding=speech.types.RecognitionConfig.LINEAR16,
+            encoding=speech.types.RecognitionConfig.AudioEncoding.LINEAR16,
             sample_rate_hertz=AUDIO_SAMPLE_RATE_HZ,
             language_code=language_code,
             speech_contexts=[speech.types.SpeechContext(phrases=hint_phrases)])

```

josecarlosphp, 2021-02-15:
It works, thank you.

nine86, 2021-02-18:
Thank you! That was super useful.

AlpenglowInd, 2021-03-03:
Thank you so much!  I was banging my head against that for hours with a fresh install of the 11/20/2020 AIY SD image (hardware is an RPi 3B and I believe a v1 AIY Voice Hat).  Worked perfectly.

LRitzdorf, 2021-05-29:
Solved for me as well. Is there a PR for this? If not, I'd be happy to make one - this is a rather important issue, since it breaks Cloud Speech functionality.

petermeester, 2021-09-10:
I had the same problem with a fresh recent VoiceKit. This solves the problem. Thank you!