google/aiyprojects-raspbian #719: Authorization Error

MihaiCristianAlphaTeam, 2020-12-06:
Hi ! I was at the `cd ~/AIY-projects-python/src/examples/voice
./assistant_library_demo.py`

part, after i got the link and pasted it into the browser i got this error:

`Error 403: access_denied
The developer hasn’t given you access to this app. It’s currently being tested and it hasn’t been verified by Google. If you think you should have access, contact the developer`

Keep in mind that i am logged in the account that i just got the api from. Is not like i am on another one...so..what can i do ?

manoj7410, 2020-12-07:
@MihaiCristianAlphaTeam Did you try to run assistant_grpc_demo too ?
Which version of voice kit are you working on ?
In which country are you running the kit ?

MihaiCristianAlphaTeam, 2020-12-07:
Yes. I ran all of them
I am on the first version because I wanted a rpi 3 inside
Romania
But I think i saw what the problem was. 
[image]

manoj7410, 2020-12-08:
@MihaiCristianAlphaTeam I just created new credentials and I am not able to reproduce this issue. Are there any specific steps that you can share?

MihaiCristianAlphaTeam, 2020-12-08:
I fixed it. I needed to publish my app....as not shown in the tutorial

closed 2020-12-08 by MihaiCristianAlphaTeam

qaz6209031, 2021-01-12:
I am having the exact same issue

qaz6209031, 2021-01-12:
I figureed it out, yeah you have to publish your app, it's weird that they don't put it in the documentation

jayquijano, 2021-01-15:
Having the same issue, how does one publish their app

bwbass, 2021-01-15:
I am in the same boat... I followed each step in the documentation for the AIY Voice Kit V2.0, but when I try to run assistant_grpc_demo.py I get:

Authorization Error
Error 403: access_denied
The developer hasn’t given you access to this app.

If the answer is publishing the app, how to I do that?

MihaiCristianAlphaTeam, 2021-01-17:
> I am in the same boat... I followed each step in the documentation for the AIY Voice Kit V2.0, but when I try to run assistant_grpc_demo.py and put the specified URL in my browser I get:
> 
> Authorization Error
> Error 403: access_denied
> The developer hasn’t given you access to this app.
> 
> If the answer is publishing the app, how to I do that?

I can't remember how i did that...but send me a screenshot of the google cloud project and maybe i will remember. You can add me on discord: AD Mihai#2499

MihaiCristianAlphaTeam, 2021-01-17:
> I am in the same boat... I followed each step in the documentation for the AIY Voice Kit V2.0, but when I try to run assistant_grpc_demo.py and put the specified URL in my browser I get: [...]

I can't remember how i did that...but send me a screenshot of the google cloud project and maybe i will remember. You can add me on discord: AD Mihai#2499

NorkaKnight, 2021-01-29:
To publish, go to the OAuth Consent Screen and under Publishing Status you can find the option to publish.

gigamegawatts, 2021-05-27:
I just ran into the same problem.  If you don't want to use the Publish App approach (is there a downside to doing this?), an alternative is to add your e-mail address as a tester.   Click OAuth Consent Screen in the left menu, then click the Add Users button under "Test Users".

I'm pretty sure that Google has changed the Cloud Console's Credentials page, since I didn't have this problem the last time I played with AIY Voice (July last year).  OK Google, update the User's Guide!

scottamain, 2021-10-11:
The credentials setup instructions are fixed to include adding your account as a test user (among other cloud console UI changes):
https://aiyprojects.withgoogle.com/voice#google-assistant--get-credentials