google/aiyprojects-raspbian #720: Vision Bonnet board not working

alpha-david, 2020-12-14:
My Vision Bonnet board is not working now. I try to figure out what is the problem
After booting up, I typed the following command and found not any aiy module being actually loaded
$ sudo systemctl stop joy_detection_demo.service
$ sudo rmmod aiy_vision
$ sudo rmmod aiy_adc
$ sudo rmmod pwm_aiy_io
$ sudo rmmod gpio_aiy_io
$ sudo rmmod aiy_io_i2c

then I typed the command and got the error like this
$ i2cdetect -y 1
Error: Could not open file `/dev/i2c-1' or `/dev/i2c/1': No such file or directory

Is it the problem of the short flexible ribbon cable? Or the entire Vision Bonnet it  broken?

manoj7410, 2020-12-16:
@alpha-david Can you confirm if you have flashed your sdcard with the latest AIY image : https://github.com/google/aiyprojects-raspbian/releases/tag/v20201120 ?

ricardodeazambuja, 2020-12-23:
I just flashed the latest image (aiyprojects-2020-11-20.img). I can't access the camera (Raspicam V2.1) when I connect it following the AIY website instructions (flat cable from RPI Zero W to AIY Bonnet and camera connected directly to the AIY Bonnet). Tried the simplest `raspistill -o test.png` and the output was ```Camera control callback  cmd=0x4f525245mmal: No data received from sensor. Check all connections, including the Sunny one on the camera board```. I was hoping this problem was fixed in the latest image... :(