google/aiyprojects-raspbian #721: Assistant Demo Not Working

snavvysnavv, 2020-12-15:
Attempting to run the Assistant Demo for Voice Kit and receive this message:

DEBUG:requests_oauthlib.oauth2_session:Generated new state DrumRzdNVsQYoQydVQ8AROxqH3uE92.
Please visit this URL to authorize this application: https://accounts.google.com/o/oauth2/auth?response_type=code&client_id=1015473307077-hpv1635utgp0nssubt7so4ejrrbsjfjb.apps.googleusercontent.com&redirect_uri=http%3A%2F%2Flocalhost%3A8080%2F&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fassistant-sdk-prototype&state=DrumRzdNVsQYoQydVQ8AROxqH3uE92&access_type=offline

I click the URL and attempt to log into my google account, and then receive an error message of "Cannot Connect to Local Host". 

Any help would be appreciated

manoj7410, 2020-12-16:
@snavvysnavv Are you logging in into the same google account from where you generated the assistant.json file ?
I just tried to execute the demo on my voice kit and was able to run it successfully. 
Can you also share the complete logs ?

MihaiCristianAlphaTeam, 2021-02-08:
Had that same issue...go to 0auth and publish the app

nmoynihan, 2021-02-24:
> Attempting to run the Assistant Demo for Voice Kit and receive this message:
>
> DEBUG:requests_oauthlib.oauth2_session:Generated new state DrumRzdNVsQYoQydVQ8AROxqH3uE92.
> Please visit this URL to [...]

The issue is that the script contains a callback to https://localhost:8080/..., meaning it should be run locally in a pi web browser. This is crazy to me, since the browser on the pi zero (Chromium) is utterly unusable.... 

I have tried opening the URL locally and it requires that I sign in to my google account, which I literally can't do because it crashes chromium. I've tried other browsers (Epiphany, Midori, Links2) and they also don't work.

Anyone have any bright ideas?