google/aiyprojects-raspbian #722: Voice Kit authorization failing

TheMontezuma, 2020-12-16:
My voice kit does not work :(
Here are the logs:

Dec 16 23:16:51 raspberrypi python3[1129]: DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token
Dec 16 23:16:51 raspberrypi python3[1129]: DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
Dec 16 23:16:51 raspberrypi python3[1129]: DEBUG:urllib3.connectionpool:https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 400 None
Dec 16 23:16:51 raspberrypi python3[1129]: Traceback (most recent call last):
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
Dec 16 23:16:51 raspberrypi python3[1129]:     main()
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 49, in main
Dec 16 23:16:51 raspberrypi python3[1129]:     language_code=args.language)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
Dec 16 23:16:51 raspberrypi python3[1129]:     super().__init__(language_code, volume_percentage)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
Dec 16 23:16:51 raspberrypi python3[1129]:     credentials = auth_helpers.get_assistant_credentials()
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
Dec 16 23:16:51 raspberrypi python3[1129]:     return _try_to_get_credentials(credentials_file)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 100, in _try_to_get_credentials
Dec 16 23:16:51 raspberrypi python3[1129]:     return _load_credentials(_ASSISTANT_CREDENTIALS)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 63, in _load_credentials
Dec 16 23:16:51 raspberrypi python3[1129]:     credentials.refresh(http_request)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/usr/local/lib/python3.7/dist-packages/google/oauth2/credentials.py", line 136, in refresh
Dec 16 23:16:51 raspberrypi python3[1129]:     self._client_secret))
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 237, in refresh_grant
Dec 16 23:16:51 raspberrypi python3[1129]:     response_data = _token_endpoint_request(request, token_uri, body)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 111, in _token_endpoint_request
Dec 16 23:16:51 raspberrypi python3[1129]:     _handle_error_response(response_body)
Dec 16 23:16:51 raspberrypi python3[1129]:   File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 61, in _handle_error_response
Dec 16 23:16:51 raspberrypi python3[1129]:     error_details, response_body)
Dec 16 23:16:51 raspberrypi python3[1129]: google.auth.exceptions.RefreshError: ('invalid_grant: Bad Request', '{\n  "error": "invalid_grant",\n  "error_description": "Bad Request"\n}')
Dec 16 23:16:51 raspberrypi systemd[1]: assistant_grpc_demo.service: Main process exited, code=exited, status=1/FAILURE
Dec 16 23:16:51 raspberrypi systemd[1]: assistant_grpc_demo.service: Failed with result 'exit-code'.
Dec 16 23:16:52 raspberrypi systemd[1]: assistant_grpc_demo.service: Service RestartSec=100ms expired, scheduling restart.
Dec 16 23:16:52 raspberrypi systemd[1]: assistant_grpc_demo.service: Scheduled restart job, restart counter is at 74.

manoj7410, 2020-12-17:
@TheMontezuma Have you followed the exact steps given at : https://aiyprojects.withgoogle.com/voice/#google-assistant--get-credentials to create the credentials?
At first run of assistant_grpc_demo.py, you would have received an URL on the terminal that you need to copy into a browser and allow the google assistant to access your google account.

TheMontezuma, 2020-12-17:
Hello Manoj,
thanks for the quick response. I setup everything about a year ago.
The last time I tested the google assistant - it was in April 2020 and it still worked.
I guess there were some changes in Google policies or something like that. 
I don't want to start from scratch with the setup. Do you have any ideas what has changed since April?
Thanks
Marcin

TheMontezuma, 2020-12-17:
` /usr/bin/python3 -u /home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py --language en-US
DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token
DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443
DEBUG:urllib3.connectionpool:https://oauth2.googleapis.com:443 "POST /token HTTP/1.1" 400 None
Traceback (most recent call last):
  File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 57, in <module>
    main()
  File "/home/pi/AIY-projects-python/src/examples/voice/assistant_grpc_demo.py", line 49, in main
    language_code=args.language)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in __init__
    super().__init__(language_code, volume_percentage)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in __init__
    credentials = auth_helpers.get_assistant_credentials()
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials
    return _try_to_get_credentials(credentials_file)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 100, in _try_to_get_credentials
    return _load_credentials(_ASSISTANT_CREDENTIALS)
  File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 63, in _load_credentials
    credentials.refresh(http_request)
  File "/usr/local/lib/python3.7/dist-packages/google/oauth2/credentials.py", line 136, in refresh
    self._client_secret))
  File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 237, in refresh_grant
    response_data = _token_endpoint_request(request, token_uri, body)
  File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 111, in _token_endpoint_request
    _handle_error_response(response_body)
  File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 61, in _handle_error_response
    error_details, response_body)
google.auth.exceptions.RefreshError: ('invalid_grant: Bad Request', '{\n  "error": "invalid_grant",\n  "error_description": "Bad Request"\n}')`

Please look at the above traces

TheMontezuma, 2020-12-17:
OK. It probably didn't work, because I haven't used the VoiceKit for more than 6 months, see:
https://blog.timekit.io/google-oauth-invalid-grant-nightmare-and-how-to-fix-it-9f4efaf1da35

I created new credentials and got it working again.

closed 2020-12-17 by TheMontezuma

cbmainz, 2021-01-15:
Hi @TheMontezuma I have the same issue I think. Did work in April, since then it was inactive and now I get the same error message. 

Did you create a new assistant.json file? I tried this but I don’t get a change to reauthorise a token or something. What steps did you take to make it work?

thanks, Christopher

TheMontezuma, 2021-01-16:
> Did you create a new assistant.json file? 

Yes.

> I tried this but I don’t get a change to reauthorise a token or something. What steps did you take to make it work?

delete ~/.cache/voice-recognizer/assistant_credentials.json

source: https://github.com/google/aiyprojects-raspbian/issues/288

cbmainz, 2021-01-17:
That helped. Thanks a lot. After hours of research and trial and error this was finally the solution.

Maybe this should be documented some were?