google/aiyprojects-raspbian #727: Voice Bonnet drivers won't install on Raspbian

WhyNine, 2021-02-08:
I am having problems installed the Voice Bonnet driver, ie it fails.

I have written the latest Raspbian to an SD card, then:

- run "echo "deb https://packages.cloud.google.com/apt aiyprojects-stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list"
- run "curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -"
- run "sudo apt update"
- run "sudo apt-get install -y aiy-voicebonnet-soundcard-dkms"

This fails with:
```
Building initial module for 5.10.11+
Error! Bad return status for module build on kernel: 5.10.11+ (armv7l)
Consult /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/make.log for more information.
dpkg: error processing package aiy-voicebonnet-soundcard-dkms (--configure):
 installed aiy-voicebonnet-soundcard-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 aiy-voicebonnet-soundcard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
The make is attached: [make.log](https://github.com/google/aiyprojects-raspbian/files/5945558/make.log)
The first error is:
```
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_init
’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:39:43: error: ‘struct snd_soc_pcm_runtime’
 has no member named ‘codec_dai’
  struct snd_soc_component *component = rtd->codec_dai->component;
                                           ^~
```
This used to work but somehow is now broken. I have tried doing an upgrade before installed the driver but no difference. I have tried full Raspbian and also Lite with the same result.

What am I doing wrong?

Please help!

builderjer, 2021-02-19:
I'm having the same issue, just a little different error.

`DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.10.11+ (armv7l)
Fri 19 Feb 22:51:23 GMT 2021
make: Entering directory '/usr/src/linux-headers-5.10.11+'
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rl6231.o
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_init’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:39:43: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  struct snd_soc_component *component = rtd->codec_dai->component;
                                           ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:48:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_MCLK,
                                  ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_hw_params’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:73:31: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_pll(rtd->codec_dai, 0, RT5645_PLL1_S_MCLK,
                               ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:80:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_PLL1, freq,
                                  ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:87:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_PLL1, freq,
                                  ^~
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.c: In function ‘is_sys_clk_from_pll’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.c:869:8: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implicit-function-declaration]
  val = snd_soc_component_read32(component, RT5645_GLB_CLK);
        ^~~~~~~~~~~~~~~~~~~~~~~~
        snd_soc_component_read
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o] Error 1
make: *** [Makefile:1808: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.11+'

I would love to use this`

CJConnor, 2021-02-22:
Hello I think I am having the same issue, I raised an issue #728 so I will close that one so there is just one issue. My make log is DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.10.11+ (armv7l)
Mon 22 Feb 14:34:07 GMT 2021
make: Entering directory '/usr/src/linux-headers-5.10.11+'
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rl6231.o
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_init’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:39:43: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  struct snd_soc_component *component = rtd->codec_dai->component;
                                           ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:48:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_MCLK,
                                  ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c: In function ‘snd_rpi_aiy_voicebonnet_hw_params’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:73:31: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_pll(rtd->codec_dai, 0, RT5645_PLL1_S_MCLK,
                               ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:80:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_PLL1, freq,
                                  ^~
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:87:34: error: ‘struct snd_soc_pcm_runtime’ has no member named ‘codec_dai’
  ret = snd_soc_dai_set_sysclk(rtd->codec_dai, RT5645_SCLK_S_PLL1, freq,
                                  ^~
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o] Error 1
make[1]: *** Waiting for unfinished jobs....
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.c: In function ‘is_sys_clk_from_pll’:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.c:869:8: error: implicit declaration of function ‘snd_soc_component_read32’; did you mean ‘snd_soc_component_read’? [-Werror=implic$
  val = snd_soc_component_read32(component, RT5645_GLB_CLK);
        ^~~~~~~~~~~~~~~~~~~~~~~~
        snd_soc_component_read
cc1: some warnings being treated as errors
make[1]: *** [scripts/Makefile.build:279: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.o] Error 1
make: *** [Makefile:1808: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.10.11+'

viraniac, 2021-02-26:
@WhyNine @CJConnor @builderjer I have created a deb file with fix for this issue. Feel free to install the same from [here](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/3.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_3.0-1+nmu1_all.deb)

WhyNine, 2021-02-27:
> 
> 
> @WhyNine @CJConnor @builderjer I have created a deb file with fix for this issue. Feel free to install the same from [here](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/3.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_3.0-1+nmu1_all.deb)

Thanks, I will give it a try in the next few days

WhyNine, 2021-03-02:
> Thanks, I will give it a try in the next few days

I tried this file. It installs without errors but the bonnet does not appear in the list of play/record devices.

viraniac, 2021-03-02:
> I tried this file. It installs without errors but the bonnet does not appear in the list of play/record devices.

That's interesting. Could you please try rebooting your kit? Also could you please share the output of `dmesg`, `lsmod` and `dtoverlay -l` command?

WhyNine, 2021-03-03:
> That's interesting. Could you please try rebooting your kit? Also could you please share the output of `dmesg`, `lsmod`, `aplay -l` and `dtoverlay -l` command?

A reboot made no difference. 

'aplay -l':
```
**** List of PLAYBACK Hardware Devices ****
card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
```

'dtoverlay -l':
`No overlays loaded`

The output of lsmod and dmesg are in the attached files.

[lsmod.log](https://github.com/google/aiyprojects-raspbian/files/6075431/lsmod.log)
[dmesg.log](https://github.com/google/aiyprojects-raspbian/files/6075433/dmesg.log)

viraniac, 2021-03-03:
> A reboot made no difference.
>
> 'aplay -l':
>
> ```
> **** List of PLAYBACK Hardware Devices ****
> card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
>   Subdevices: [...]

What version of voice kit do you have? Is it v1 or v2/voice bonnet? Also which version of pi are you using?

CJConnor, 2021-03-03:
> > > That's interesting. Could you please try rebooting your kit? Also could you please share the output of `dmesg`, `lsmod`, `aplay -l` and `dtoverlay -l` command?
> > 
> > A reboot made no difference.
> > 
> > 'aplay -l':
> > 
> > ```
> > **** List of PLAYBACK Hardware Devices ****
> > card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
> >   Subdevices: 8/8
> >   Subdevice #0: subdevice #0
> >   Subdevice #1: subdevice #1
> >   Subdevice #2: subdevice #2
> >   Subdevice #3: subdevice #3
> >   Subdevice #4: subdevice #4
> >   Subdevice #5: subdevice #5
> >   Subdevice #6: subdevice #6
> >   Subdevice #7: subdevice #7
> > ```
> > 
> > 'dtoverlay -l':
> > `No overlays loaded`
> > 
> > The output of lsmod and dmesg are in the attached files.
> > 
> > [lsmod.log](https://github.com/google/aiyprojects-raspbian/files/6075431/lsmod.log)
> > [dmesg.log](https://github.com/google/aiyprojects-raspbian/files/6075433/dmesg.log)
> 
> What version of voice kit do you have? Is it v1 or v2/voice bonnet? Also which version of pi are you using? These drivers are only needed for v2 kit. 

Yes I tried the .deb  file and I experienced the same issues. I am using the v2 and a raspberry pi 3a+

viraniac, 2021-03-03:
> Yes I tried the .deb file and I experienced the same issues. I am using the v2 and a raspberry pi 3a+

My v2 is connected to the pi zero that came with it and that seems to be working fine. Let me dissassemble it and check with other v3 model b as well. I don't have a 3a+.

But you are sure its v2 right and not t[his one](https://aiyprojects.withgoogle.com/static/images/voice-v1/assembly/hardware-5.jpg)?

CJConnor, 2021-03-03:
> > > > > That's interesting. Could you please try rebooting your kit? Also could you please share the output of `dmesg`, `lsmod`, `aplay -l` and `dtoverlay -l` command?
> > > > 
> > > > 
> > > > A reboot made no difference.
> > > > 'aplay -l':
> > > > ```
> > > > **** List of PLAYBACK Hardware Devices ****
> > > > card 0: Headphones [bcm2835 Headphones], device 0: bcm2835 Headphones [bcm2835 Headphones]
> > > >   Subdevices: 8/8
> > > >   Subdevice #0: subdevice #0
> > > >   Subdevice #1: subdevice #1
> > > >   Subdevice #2: subdevice #2
> > > >   Subdevice #3: subdevice #3
> > > >   Subdevice #4: subdevice #4
> > > >   Subdevice #5: subdevice #5
> > > >   Subdevice #6: subdevice #6
> > > >   Subdevice #7: subdevice #7
> > > > ```
> > > > 
> > > > 
> > > > 'dtoverlay -l':
> > > > `No overlays loaded`
> > > > The output of lsmod and dmesg are in the attached files.
> > > > [lsmod.log](https://github.com/google/aiyprojects-raspbian/files/6075431/lsmod.log)
> > > > [dmesg.log](https://github.com/google/aiyprojects-raspbian/files/6075433/dmesg.log)
> > > 
> > > 
> > > What version of voice kit do you have? Is it v1 or v2/voice bonnet? Also which version of pi are you using? These drivers are only needed for v2 kit.
> > 
> > 
> > Yes I tried the .deb file and I experienced the same issues. I am using the v2 and a raspberry pi 3a+
> 
> My v2 is connected to the pi zero that came with it and that seems to be working fine. Let me dissassemble it and check with other 3b as well. I don't have a 3a+.
> 
> But you are sure its v2 right and not [this one](https://aiyprojects.withgoogle.com/static/images/voice-v1/assembly/hardware-5.jpg)?

Yes definitely v2, and I had got it working before on the pi 3a+ so I know it should work but this time when I tried to install it, it just wouldn't work

viraniac, 2021-03-03:
> Yes definitely v2, and I had got it working before on the pi 3a+ so I know it should work but this time when I tried to install it, it just wouldn't work

OK. I tried and almost broke my pi zero pin trying to disassemble the bonnet. The pins have gone weak because of the number of times I have disassembled it in the past. As I didn't wanted to break the same, I am not going to disassemble it again.

Just to check if your hat is getting detected, can you share the output of `cat /sys/firmware/devicetree/base/hat/product`. Also I advice you to once try to see if your bonnet is working on pi zero just to rule out that the bonnet is not faulty.

WhyNine, 2021-03-03:
I am using a Pi4. I know the Voice Bonnet is working as it is detected correctly when I use the Google Raspbian image.

viraniac, 2021-03-03:
> I am using a Pi4. I know the Voice Bonnet is working as it is detected correctly when I use the Google Raspbian image.

I never said that the voice bonnet won't work with Pi4. I have used it myself back in october last year. I just want to make sure you have voice bonnet.

WhyNine, 2021-03-03:
> 
> 
> > I am using a Pi4. I know the Voice Bonnet is working as it is detected correctly when I use the Google Raspbian image.
> 
> I never said that the voice bonnet won't work with Pi4. I have used it myself back in october last year. I just want to make sure you have voice bonnet.

Yes, it's definitely a bonnet, ie v2.

viraniac, 2021-03-03:
> Yes, it's definitely a bonnet, ie v2.

Could you please share the output of `cat /sys/firmware/devicetree/base/hat/product` command. This should work regardless of whether you have any drivers for bonnet installed or not.

WhyNine, 2021-03-03:
The result is `AIY VoiceBonnet`

viraniac, 2021-03-03:
Ok. Thats good.

Could you install leds-ktd202x-dkms and aiy-dkms packages and then share
the output of lsmod command? Those packages are available in google's
package repository.

This is to check if other modules are working correctly.

WhyNine, 2021-03-04:
> 
> Ok. Thats good. Could you install leds-ktd202x-dkms and aiy-dkms packages and then share the output of lsmod command? Those packages are available in google's package repository. This is to check if other modules are working correctly.
> […](#)

I installed these packages ok and rebooted, but no difference.

viraniac, 2021-03-04:
> I installed these packages ok and rebooted, but no difference.

Those are drivers to get the button and leds working. They won't make the card to appear in the aplay list, but I wanted to check if they are getting loaded automatically. Hence it will be great if you can share the `lsmod` output.

Also have you made any modifications to the `/boot/config.txt` file? Can you share your `/boot/config.txt` file here?

WhyNine, 2021-03-04:
The output of lsmod is here:
[lsmod2.log](https://github.com/google/aiyprojects-raspbian/files/6083463/lsmod2.log)

And the config.txt is here:
[config.txt](https://github.com/google/aiyprojects-raspbian/files/6083464/config.txt)

viraniac, 2021-03-04:
> The output of lsmod is here:
> [lsmod2.log](https://github.com/google/aiyprojects-raspbian/files/6083463/lsmod2.log)
> 
> And the config.txt is here:
> [config.txt](https://github.com/google/aiyprojects-raspbian/files/6083464/config.txt)

Ok... those got loaded correctly

What happens when you manually load the snd_aiy_voicebonnet module using `sudo modprobe snd_aiy_voicebonnet`. Does the card then appear in the output of `aplay -l`

WhyNine, 2021-03-04:
It says "modprobe: ERROR: could not insert 'snd_aiy_voicebonnet': Invalid argument"

viraniac, 2021-03-04:
> It says "modprobe: ERROR: could not insert 'snd_aiy_voicebonnet': Invalid argument"

Thats progress. That means its trying to initialize some value for the card and that, for some reason, is not recognized by the kernel.

Now could you please share the output of `sudo modprobe -v snd_aiy_voicebonnet` and `sudo strace modprobe -v snd_aiy_voicebonnet`

WhyNine, 2021-03-04:
sudo modprobe -v snd_aiy_voicebonnet
insmod /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko 
modprobe: ERROR: could not insert 'snd_aiy_voicebonnet': Invalid argument

sudo strace modprobe -v snd_aiy_voicebonnet 2> log.txt
insmod /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko
[log.txt](https://github.com/google/aiyprojects-raspbian/files/6084318/log.txt)

The output of dmesg is:
[dmesg.log](https://github.com/google/aiyprojects-raspbian/files/6084321/dmesg.log)

viraniac, 2021-03-04:
> sudo modprobe -v snd_aiy_voicebonnet
> insmod /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko
> modprobe: ERROR: could not insert 'snd_aiy_voicebonnet': Invalid argument
>
> sudo [...]

There is something weird going on here. When I run `sudo modprobe -v snd_aiy_voicebonnet`, it runs insmod on 4 files for me with snd-aiy-voicebonnet.ko being the last one. Something like this

```
insmod /lib/modules/5.10.11-v7l+/kernel/drivers/base/regmap/regmap-i2c.ko
insmod /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rl6231.ko
insmod /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rt5645.ko
insmod /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko
```

For you its only showing the last one and then failing. Can you check that the files that are listed in front of insmod above exists for you.

If the files exists, can you run the above 4 insmod commands yourself and share the output. You have to run them as root.

WhyNine, 2021-03-04:
All 4 files exist but all except regmap-i2c.ko have a length of 0.

```
pi@raspberrypi:~ $ sudo insmod /lib/modules/5.10.11-v7l+/kernel/drivers/base/regmap/regmap-i2c.ko
insmod: ERROR: could not insert module /lib/modules/5.10.11-v7l+/kernel/drivers/base/regmap/regmap-i2c.ko: File exists
pi@raspberrypi:~ $ sudo insmod /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rl6231.ko
insmod: ERROR: could not insert module /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rl6231.ko: Invalid parameters
pi@raspberrypi:~ $ sudo insmod /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rt5645.ko
insmod: ERROR: could not insert module /lib/modules/5.10.11-v7l+/kernel/sound/soc/codecs/rt5645.ko: Invalid parameters
pi@raspberrypi:~ $ sudo insmod /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko
insmod: ERROR: could not insert module /lib/modules/5.10.11-v7l+/kernel/sound/drivers/snd-aiy-voicebonnet.ko: Invalid parameters
```

coopht, 2021-03-04:
Here is how I fixed the build: put attached patches 
[rt5645.patch.txt](https://github.com/google/aiyprojects-raspbian/files/6084686/rt5645.patch.txt)
[voice-bonnet.patch.txt](https://github.com/google/aiyprojects-raspbian/files/6084687/voice-bonnet.patch.txt)
into  `/usr/src/aiy-voicebonnet-soundcard-3.0/patches ` directory,
then edit `/usr/src/aiy-voicebonnet-soundcard-3.0/dkms.conf` file and add patches filenames there, 
I added PATCH[0] and PATCH[1] below:
```
BUILT_MODULE_NAME[0]="snd-aiy-voicebonnet"
DEST_MODULE_LOCATION[0]="/kernel/sound/drivers"
PATCH[0]=voice-bonnet.patch
BUILT_MODULE_NAME[1]="rt5645"
DEST_MODULE_LOCATION[1]="/kernel/sound/soc/codecs"
PATCH[1]=rt5645.patch
```
After all that, just run `sudo apt-get install -y aiy-voicebonnet-soundcard-dkms` again. 
It worked for me rpi3 5.10.11-v7+ kernel.
Hope it helps.

viraniac, 2021-03-04:
> All 4 files exist but all except regmap-i2c.ko have a length of 0.
>
> ```
> pi@raspberrypi:~ $ sudo insmod /lib/modules/5.10.11-v7l+/kernel/drivers/base/regmap/regmap-i2c.ko
> insmod: ERROR: could not [...]

If those files are of 0 bytes then your installation is corrupt. try removing and reinstalling the deb file.

viraniac, 2021-03-04:
> Here is how I fixed the build: put attached patches
> [rt5645.patch.txt](https://github.com/google/aiyprojects-raspbian/files/6084686/rt5645.patch.txt) [...]

I have made similar changes and created the deb file of the same to make it easier for others. The code is hosted [here](https://github.com/viraniac/aiy-voicebonnet-soundcard-dkms/releases/download/3.0-1%2Bnmu1/aiy-voicebonnet-soundcard-dkms_3.0-1+nmu1_all.deb) in my git repository. Apart from changes that you have made, I have also synchronised the rl6232 and rt5645 files with the upstream kernel repository as they don't contain any changes specially made for the voice bonnet.

WhyNine, 2021-03-04:
Well, I deleted those 3 files and tried installing the deb file again, but it says that `aiy-voicebonnet-soundcard-dkms is already the newest version (3.0-1+nmu1)`.

viraniac, 2021-03-04:
> Well, I deleted those 3 files and tried installing the deb file again, but it says that `aiy-voicebonnet-soundcard-dkms is already the newest version (3.0-1+nmu1)`.

I guess you misunderstood me. Uninstall the deb and install again.

WhyNine, 2021-03-04:
That's done it, many thanks. But what were the critical steps so that I can reproduce it?

viraniac, 2021-03-04:
> That's done it, many thanks. But what were the critical steps so that I can reproduce it?

Reproduce the issue or reproduce the working environment? 

If you are talking about working environment. removing existing deb from google and installing mine fixes the dkms errors. 

if you are talking about the 0-byte file that you had, I have no clue. Looks like some kind of storage issue. either your storage was full or may be you had a power failure or something like that. I have no idea.

WhyNine, 2021-03-04:
Ok, I'll have a play with a clean install and try to isolate the correct steps.

viraniac, 2021-03-04:
> Ok, I'll have a play with a clean install and try to isolate the correct steps.

Well if you just needs some commands to have things working, you can refer my comment [here](https://github.com/google/aiyprojects-raspbian/issues/697#issuecomment-703084054). Just use the latest deb file from my repository instead of the one I mentioned there.

viraniac, 2021-03-04:
> Ok, I'll have a play with a clean install and try to isolate the correct steps.

What you were following in your first post, were the right steps. Its just latest raspbian kernel is not compatible with the deb file that google has in their repository. And Google generally doesn't update things prompty when it breaks. Hence I try to provide fixed files for others to make it easier for them until official update is released by Google

WhyNine, 2021-03-04:
I have just tried with a clean install again. These are the steps I followed:
```
echo "deb https://packages.cloud.google.com/apt aiyprojects-stable main" | sudo tee /etc/apt/sources.list.d/aiyprojects.list
curl https://packages.cloud.google.com/apt/doc/apt-key.gpg | sudo apt-key add -
sudo apt update
sudo apt upgrade
sudo reboot         *****this is very important*****
sudo apt install ./aiy-voicebonnet-soundcard-dkms_3.0-1+nmu1_all.deb
```

The only difference to what I did before, I think, is that I may not have done the reboot after the upgrade. But certainly it worked this time.

dmitriykovalev, 2021-04-09:
We've updated sound driver and put all source code to https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/drivers.
There is a new [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20210402) as well.

iamaturtle113, 2021-04-15:
> I have just tried with a clean install again. These are the steps I followed:
>
> ```
> echo "deb https://packages.cloud.google.com/apt aiyprojects-stable main" | sudo tee [...]

My AIY case, install failed.

> We've updated sound driver and put all source code to https://github.com/google/aiyprojects-raspbian/tree/aiyprojects/drivers.
> There is a new [SD card image](https://github.com/google/aiyprojects-raspbian/releases/tag/v20210402) as well.

I tried 
sudo python setup.py install
but got error.
```

Searching for google-assistant-library>=1.0.1
Reading https://pypi.python.org/simple/google-assistant-library/
No local packages or working download links found for google-assistant-library>=1.0.1
error: Could not find suitable distribution for Requirement.parse('google-assistant-library>=1.0.1')
```

viraniac, 2021-04-18:
> My AIY case, install failed.

You don't need to install my package now that google has updated their own. Install it directly from their repository as mentioned in Hacking.md file.
> I tried
> sudo python setup.py install
> but got error.
>
> ```
>
> Searching for google-assistant-library>=1.0.1
> Reading https://pypi.python.org/simple/google-assistant-library/
> No local packages or working [...]

Have you installed the `aiy-python-wheels` package before trying to run setup.py?