google/aiyprojects-raspbian #728: Sub-process /usr/bin/dpkg returned an error code (1)

CJConnor, 2021-02-09:
Hello, 

I am working on a project involving my google aiy v2 voice bonnet and my raspberry pi 3 a+, but every time I try to install the google aiy voicebonnet driver I get the following resulting in the error code in the title. I am using the desktop version of the raspberry pi 32 bit OS, I am new to the google aiy v2 kit and the raspberry pi 3 a+ so I am unsure what this error could be. Any help would be greatly appreciated :)

I followed the instructions in HACKING.md to install the aiy-voicebonnet-soundcard-dkms driver, but when I do I get this error.

sudo apt-get install -y aiy-voicebonnet-soundcard-dkms
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following additional packages will be installed:
  dkms raspberrypi-kernel-headers
Suggested packages:
  python3-apport menu
The following NEW packages will be installed:
  aiy-voicebonnet-soundcard-dkms dkms raspberrypi-kernel-headers
0 upgraded, 3 newly installed, 0 to remove and 0 not upgraded.
Need to get 27.7 MB of archives.
After this operation, 180 MB of additional disk space will be used.
Get:1 http://archive.raspberrypi.org/debian buster/main armhf raspberrypi-kernel-headers armhf 1.20210201-1 [27.6 MB]
Get:3 https://packages.cloud.google.com/apt aiyprojects-stable/main armhf aiy-voicebonnet-soundcard-dkms all 3.0-1 [32.9 kB]
Get:2 http://www.mirrorservice.org/sites/archive.raspbian.org/raspbian buster/main armhf dkms all 2.6.1-4 [74.4 kB]
Fetched 27.7 MB in 14s (1,973 kB/s)
Selecting previously unselected package raspberrypi-kernel-headers.
(Reading database ... 163703 files and directories currently installed.)
Preparing to unpack .../raspberrypi-kernel-headers_1.20210201-1_armhf.deb ...
Unpacking raspberrypi-kernel-headers (1.20210201-1) ...
Selecting previously unselected package dkms.
Preparing to unpack .../archives/dkms_2.6.1-4_all.deb ...
Unpacking dkms (2.6.1-4) ...
Selecting previously unselected package aiy-voicebonnet-soundcard-dkms.
Preparing to unpack .../aiy-voicebonnet-soundcard-dkms_3.0-1_all.deb ...
Unpacking aiy-voicebonnet-soundcard-dkms (3.0-1) ...
Setting up dkms (2.6.1-4) ...
Setting up raspberrypi-kernel-headers (1.20210201-1) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.11-v8+
Setting up aiy-voicebonnet-soundcard-dkms (3.0-1) ...
Loading new aiy-voicebonnet-soundcard-3.0 DKMS files...
It is likely that 5.10.11-v7+ belongs to a chroot's host
Building for 5.10.11+, 5.10.11-v7+, 5.10.11-v7l+ and 5.10.11-v8+
Building initial module for 5.10.11+
Error! Bad return status for module build on kernel: 5.10.11+ (armv7l)
Consult /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/make.log for more information.
dpkg: error processing package aiy-voicebonnet-soundcard-dkms (--configure):
 installed aiy-voicebonnet-soundcard-dkms package post-installation script subprocess returned error exit status 10
Processing triggers for man-db (2.8.5-2) ...
Errors were encountered while processing:
 aiy-voicebonnet-soundcard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

WhyNine, 2021-02-20:
Is this the same error as I reported on issue #727 ? It looks like it may be. What is in the make.log file?

CJConnor, 2021-02-22:
duplicate issue of #727