google/aiyprojects-raspbian #734: Voice kit seemingly works... just very quiet

abwins, 2021-04-28:
This is my first time using the voice kit V2, and I'm using it for a magic mirror voice assistant.
I believe I have the voice bonnet assembled correctly and I’m using it with a raspberry pi 3 b+ instead of a zero (which hasn’t seemed to be a problem with references I've seen). I also believe I set up the voice assistant correctly. When I test the audio it seems to be making noises, just VERY quietly, almost like a buzzing noise. I can hear its saying something, I just cant make out what because it’s so quiet. I’ve tried both responding yes and no to the test in regards to it working properly. When I get to the step where I record myself it plays the recording at full volume and I can clearly tell what was recorded. I’ll start it when me and my roommate are in the middle of a conversation and it will play back what we’ve said at full volume. I've also checked the volume settings and it's at max volume.
Once I get to the step of testing the google assistant, I’ve had multiple issues. The main issue I’ve had is that it will hear me ask a question, such as “what’s the weather?” but the answer is quiet just like before. On screen everything looks like it’s going fine, but the volume is insanely quiet. Last night I’ve had issues where it would start to only record the end of my questions and then none of them. It seemed to stop working more and more after each test. Since then however I had issues with magic mirror and had to wipe the pi and start over, using the AIY projects image, but as of this morning the issue I’m still having is that it’ll hear me just fine, but the speaker is still quiet. 
I guess my question is what could possibly be wrong? Is there something I could have possibly missed, or would this be an issue with the voice bonnet/speaker? Any suggestion would be extremely helpful!
sorry for the lengthy explanation but I wanted to be thorough in the steps I've taken so far.

manoj7410, 2021-06-02:
@abwins Can you ask your assistant 'set the volume at 200%' and then try the demo ?

vorburger, 2021-08-20:
@abwins have you ran `alsamixer` (SSH into it) and double checked that's not set to very low volume?