google/aiyprojects-raspbian #741: AIY python package version

sentairanger, 2021-07-22:
Hello I was wondering how do I find the version of modules like aiy.board or aiy.cloudspeech. I tried using aiy.board.__version__ with no luck. I ask this because when I ran pip freeze > requirements.txt I couldn't find aiy at all except for the GitHub website reference. So hopefully anyone can resolve this.

manoj7410, 2021-07-23:
The only pip package that I see is : AIY-projects-python and not the individual packages. Are you able to find any other packages as well?
I get the latest packages from https://github.com/google/aiyprojects-raspbian/blob/aiyprojects/HACKING.md

sentairanger, 2021-07-23:
@manoj7410 Correct that's the only one I see. I just went to my voice kit and I got a version number. I am closing this now. Thank you for helping.

closed 2021-07-23 by sentairanger