google/aiyprojects-raspbian #742: GET request in Robot framework using AWS credentials

Saruto85, 2021-08-09:
Hello All,
I'm unable to send authorization with a get request to the end point through robot framework. I succeeded in sending a get request using AWS authorization using Postman. But when I took the signed header and sent it as a dictionary in robot framework it gave me this error:
 Create session    mysession    https://url
    ${headers}=    Create Dictionary    Authorization=AWS4-HMAC-SHA256 Credential=, SignedHeaders=content-type;host;x-amz-date, Signature=    X-Amz-Date=20210809T065038Z    Content_Type=application/json
    ${response}=    Get On Session    mysession    url=${newActCode}    headers=${headers}
HTTPError: 403 Client Error: Forbidden for url: https://

Please help, I'm really stuck.
#robot frame work #aws authorization

manoj7410, 2021-08-09:
@Saruto85 Is this related to Google AIY kits ?

Saruto85, 2021-08-09:
@manoj7410 I don't know. I am using API testing on Robot framework and requests library. And I want to send a request but I must send AWS credentials with it? What do you mean by Google AIY kits?

manoj7410, 2021-08-09:
@Saruto85 The repo where you are posting this issue is specifically meant for Google AIY projects. Please see : https://github.com/google/aiyprojects-raspbian

I believe this is not the right place to post this query.

Saruto85, 2021-08-09:
@manoj7410 Oh, I'm sorry. I didn't notice. Where can I post it? thanks