google/aiyprojects-raspbian #744: AYI-vision:  Unit joy_dection_demo.service could not be found

ttl441, 2021-09-07:
Hi everyone
I need some help.  After the assembly of vision kit 1.1,  I can not have the camera working.   When I checked for joy_detection service status, I got the error message " Unit joy_detection_demo service count not be found.   what software version should i upgrade to get all the necessary to run this vision kit?

ttl441, 2021-09-07:
I used the latest image in April 2021.  I still see the same issue.

manoj7410, 2021-09-07:
@ttl441 Do you see a green LED glowing on your kit's front after a fresh bootup?

ttl441, 2021-09-10:
No - I do not.  I tested the camera is bad ( using raspberry pi 4) and replace with the good camera ) and till it does not work.
How do we test vision bonnet to see it is a good board?