google/aiyprojects-raspbian #745: AIY-voice: problems encountered by assistant_grpc_demo.py after some time of use

Erra-Ernesto, 2021-09-23:
Hi everyone, I have two different problems:

1) the voice/assistant_grpc_demo.py application works initially without problems. However, after a few hours that I have left the device on and running, the assistant_grpc_demo.py application stops working correctly. The button remains lit in bright red, but if I press it it no longer turns red and it no longer responds to my voice commands (as if it didn't listen to me at all).

2) after 7 days of use, the application gives me this error:
google.auth.exceptions.RefreshError: ('invalid_grant: Token has been expired or revoked.', '{\n  "error": "invalid_grant",\n  "error_description": "Token has been expired or revoked."\n}')
I also tried to create a new OAuth 2.0 credential with a new key that I went to overwrite in the assistant.json file but the problem persists.

Thanks in advance for an answer to these two annoying problems.

Hernane007, 2022-08-23:
Hi, my problem is similar, you need help me ? 

Please, I'm trying to perform the step to authorize the API credential, so I can use the demo assistant, but I can't find the url for authorization, the following error appears:

pi@raspberrypi:~/AIY-voice-kit-python $ src/examples/voice/assistant_grpc_demo.py
DEBUG:google.auth.transport.requests:Making request: POST https://oauth2.googleapis.com/token DEBUG:urllib3.connectionpool:Starting new HTTPS connection (1): oauth2.googleapis.com:443 DEBUG:urllib3.connectionpool:[https://oauth2.googleapis.com:443](https://oauth2.googleapis.com/) "POST /token HTTP/1.1" 400 None Traceback (most recent call last): File "src/examples/voice/assistant_grpc_demo.py", line 57, in main() File "src/examples/voice/assistant_grpc_demo.py", line 49, in main language_code=args.language) File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 258, in init super().init(language_code, volume_percentage) File "/home/pi/AIY-projects-python/src/aiy/assistant/grpc.py", line 94, in init credentials = auth_helpers.get_assistant_credentials() File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 151, in get_assistant_credentials return _try_to_get_credentials(credentials_file) File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 100, in _try_to_get_credentials return _load_credentials(_ASSISTANT_CREDENTIALS) File "/home/pi/AIY-projects-python/src/aiy/assistant/auth_helpers.py", line 63, in _load_credentials credentials.refresh(http_request) File "/usr/local/lib/python3.7/dist-packages/google/oauth2/credentials.py", line 206, in refresh self._scopes, File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 248, in refresh_grant response_data = _token_endpoint_request(request, token_uri, body) File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 124, in _token_endpoint_request _handle_error_response(response_body) File "/usr/local/lib/python3.7/dist-packages/google/oauth2/_client.py", line 60, in _handle_error_response raise exceptions.RefreshError(error_details, response_body) google.auth.exceptions.RefreshError: ('invalid_grant: Bad Request', '{\n "error": "invalid_grant",\n "error_description": "Bad Request"\n}')

Erra-Ernesto, 2022-08-24:
> It has been a while since I solved this problem. I don't remember very
> well, but if I'm not mistaken it was enough to download the service key
> from the Google project and then mark the Google project as "product"
> instead of "test". I don't remember very well, but in the settings of the
> google platform you can switch from the testing phase to production, and in
> this way the key does not expire after about a week. I'm sorry I'm not of
> more help.
>
> Il giorno mer 24 ago 2022 alle ore 01:28 Hernane007 <

I found these notes that I had written down. Hope they can help:

After 7 days of use, the token expires and this error message is returned:

google.auth.exceptions.RefreshError: ('invalid_grant: Token has been
expired or revoked.', '{\ n "error": "invalid_grant", \ n
"error_description": "Token has been expired or revoked." \ n } ')

At the moment to solve it you need to create a new OAuth 2.0 credential,
download the new key and copy the contents of the key to the file
assistant.json

after which you need to remove the file
~ / .cache / voice-recognizer / assistant_credentials.json

Then run the application again.

When rerun for the first time, the application asks you to copy a url link
in the browser to obtain the right to access the application. If access is
refused, it is likely that the account you are using is not among the
application testers.