google/aiyprojects-raspbian #749: aiy-python-wheels fails to get installed on raspbian buster

tarzanek, 2021-11-10:
as per HACKING.md trying to get wheels installed:
```
pi@raspberrypi:~ $ sudo apt-get install -y aiy-python-wheels
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
aiy-python-wheels is already the newest version (1.4-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Setting up aiy-python-wheels (1.4-1) ...
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: protobuf-3.14.0-cp37-cp37m-linux_armv7l.whl is not a supported wheel on this platform.
dpkg: error processing package aiy-python-wheels (--configure):
 installed aiy-python-wheels package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 aiy-python-wheels
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.63-v7l+ #1459 SMP Wed Oct 6 16:41:57 BST 2021 armv7l GNU/Linux
```

can support for v7l+ be added?

tarzanek, 2021-11-10:
@dmitriykovalev pretty please? :-)

LRitzdorf, 2021-11-12:
I'm currently working through my own reinstall on top of Raspberry Pi OS Bullseye, and I ran into the same issue. I was able to resolve it using the instructions at https://github.com/google/aiyprojects-raspbian/issues/526#issuecomment-491370978.

tarzanek, 2021-11-14:
seems skipping that wheel looks safe enough @LRitzdorf 
since:
```
root@raspberrypi:~# pip3 list | grep proto
googleapis-common-protos 1.53.0
proto-plus               1.19.8
protobuf                 3.19.1
```
and above is trying to install 3.14

LRitzdorf, 2021-11-14:
Hm, interesting. I thought Google installed their own optimized `protobuf` fork, but I'm not certain of that.
Also, would you mind letting me know how audio goes, over in #752? It sounds like we're both reinstalling on top of the base Raspbian image, so I'm curious to see if you run into the same issue I did.