google/aiyprojects-raspbian #755: 500 error code

Wa200300, 2021-12-29:
I used the latest .img.xz file from GitHub and followed all the steps from https://aiyprojects.withgoogle.com/voice-v1 exactly as explained, however, for example, after running assistant_library_demo.py and opening the URL and logging in, I received an 500 error code. I have been unable to run any demos no matter what I have tried, help to fix this would be appreciated!

manoj7410, 2021-12-29:
Can you please paste the complete error here ?
And in which country are you trying to run the voice kit?

Wa200300, 2021-12-29:
I am trying to run the voice kit in Finland.

First, I run the assistant_library_demo.py file and open the given URL.
[image]

Afterward, I log in and press continue after being told that Google hasn't verified the app.
[image]

After pressing continue, I am told something went wrong.
[image]

Finally, I press next and there is a 500 error code displayed.
[image]

kateprocter, 2022-01-19:
I also had this problem with the latest image and resolved it by running updates and using a more recent version of chromium to log in to google.