google/aiyprojects-raspbian #761: ost-installation script subprocess returned error exit status 6

butteryak, 2022-10-17:
getting an error why trying to install aiy-voicebonnet-soundcard-dkms

sudo apt-get install -y aiy-voicebonnet-soundcard-dkms

any help would be appreciate, I know possibly there have been some changes on the aiy side that may no longer be compatable...also changes to rasbperry OS.  

Building for 5.15.61+, 5.15.61-v7+, 5.15.61-v7l+ and 5.15.61-v8+
Building initial module for 5.15.61+
Error! Application of patch voice-bonnet.patch failed.
Check /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/ for more information.
dpkg: error processing package aiy-voicebonnet-soundcard-dkms (--configure):
 installed aiy-voicebonnet-soundcard-dkms package post-installation script subprocess returned error exit status 6
Errors were encountered while processing:
 aiy-voicebonnet-soundcard-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

butteryak, 2022-10-17:
DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.15.61+ (armv6l)
Mon 17 Oct 16:47:58 BST 2022
make: Entering directory '/usr/src/linux-headers-5.15.61+'
  CC [M]  /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o
In file included from /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.c:25:
/var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/rt5645.h:12:10: fatal error: sound/rt5645.h: No such file or directory
   12 | #include <sound/rt5645.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.build:289: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build/snd-aiy-voicebonnet.o] Error 1
make: *** [Makefile:1898: /var/lib/dkms/aiy-voicebonnet-soundcard/3.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.15.61+'

denics, 2022-11-06:
Here you can find up to date drivers: https://github.com/google/aiyprojects-raspbian/issues/697

builderjer, 2022-12-08:
> DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.15.61+ (armv6l) Mon 17 Oct 16:47:58 BST 2022 make: Entering directory '/usr/src/linux-headers-5.15.61+' CC [M] [...]

> DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.15.61+ (armv6l) Mon 17 Oct 16:47:58 BST 2022 make: Entering directory '/usr/src/linux-headers-5.15.61+' CC [M] [...]

> DKMS make.log for aiy-voicebonnet-soundcard-3.0 for kernel 5.15.61+ (armv6l) Mon 17 Oct 16:47:58 BST 2022 make: Entering directory '/usr/src/linux-headers-5.15.61+' CC [M] [...]

I am having the same issue, did @denics solution work for you?